NVIDIAGameWorks/PhysX #3: Empty expression statement has no effect

salamanderrake, 2018-12-21:
Fails to compiles on Ubuntu 18.04 with clang-8

```
error: empty expression statement has no effect; remove unnecessary ';' to silence this
      warning
```

AlesBorovicka, 2018-12-21:
Hi,
 I dont see clang 8 release available yet. It should compile fine with up to clang 7. We will update to clang 8 in later updates.

salamanderrake, 2018-12-21:
Yes, you are correct, I don't know what happened between the clang 7 and 8, but it compiles now and the demos run, minus one cavete, a dependenct on glut3 thats not handled by cmake, it should warn that the demos depend on glut3, I ended up installing glut3 and it works fine.

AlesBorovicka, 2018-12-21:
I see, will try to take a look at that, thanks for the report

closed 2018-12-21 by salamanderrake