NVIDIAGameWorks/PhysX #4: Regarding Pull requests

amoravanszky, 2018-12-21:
Hi there,

our legal counsel has advised us to put a CLA in place before accepting pull requests.  We are working on sorting this out but due to the impending holidays it won't happen until next year.  

Of course if you find and fix issues there is no harm in queuing up the fixes already so they are not forgotten about.  

Another thing to be aware of is that our internal day to day development happens in perforce and we will continue pushing periodic 'releases' over to github that have passed our internal testing process.  So even when we start to take pull requests they will not manifest on github immediately, but will get merged to perforce first and then appear here later as part of the next versioned update.

--Adam

closed 2018-12-21 by amoravanszky

Jamim, 2018-12-21:
> So even when we start to take pull requests they will not manifest on github immediately, but will get merged to perforce first and then appear here later as part of the next versioned update.

I believe that this approach is not so convenient and friendly to contributors :thinking:

amoravanszky, 2019-02-28:
It makes sense to me that engaged contributors would like to work differently but I cannot easily change our internal development process.  Note that between our 4.0 github release in December and today we have received 9 github pull requests which were as far as I can remember all good changes, and a big thank you goes out to the respective contributors.  We have pulled them into mainline and will be pushing them back out in the next update soon.  

However in the same time our internal dev branch has seen well over 200 checkins.  So its safe to say that by far the majority of work still happens internally and we need to keep that process as streamlined as possible.  Also I am not sure if everyone would appreciate that volume of daily churn.  I think periodic tested releases are more convenient for many people.

ValZapod, 2019-09-03:
@amoravanszky Really you should consider preserving all commit history instead of squashing the commits. It is very uncomfortable. All regression tests are not working entirely.