NVIDIAGameWorks/PhysX #8: Error when running physx/generate_projects.sh

arrfou90, 2018-12-21:
When excuting the shell script **generate_projects.sh**, I go this undefined error. 

> 
> Using preset xml: /home/nvidia/PhysX/physx/buildtools/presets/public/linux-aarch64.xml
> Target platform: linuxAarch64 using compiler: gcc
> PM_CMakeModules_PATH: /home/nvidia/PhysX/physx/../externals/cmakemodules
> PM_PATHS: /home/nvidia/PhysX/physx/../externals/opengl-linux
> Cmake: cmake
> sh: 1: cmake: not found
> sh: 1: cmake: not found
> sh: 1: cmake: not found
> sh: 1: cmake: not found
> 
After debugging, I figuerd out the error was becuase I did not have cmake installed in my system. Also, notice that for compiling CMake 3.12 or higher is required!

salamanderrake, 2018-12-21:
It works fine with cmake 3.10 on my system, I just changed this file physx/compiler/public/CMakeLists.txt as seen below with the git diff/patch

```

diff --git a/physx/compiler/public/CMakeLists.txt b/physx/compiler/public/CMakeLists.txt
index a3f5731..ebce46d 100644
--- a/physx/compiler/public/CMakeLists.txt
+++ b/physx/compiler/public/CMakeLists.txt
@@ -25,7 +25,7 @@
 ##
 ## Copyright (c) 2018 NVIDIA Corporation. All rights reserved.
 
-cmake_minimum_required(VERSION 3.12)
+cmake_minimum_required(VERSION 3.10)
 
 project(PhysXSDK C CXX)
 
@@ -163,4 +163,4 @@ IF(PX_BUILDPUBLICSAMPLES)
                SET_PROPERTY(TARGET SampleRenderer PROPERTY FOLDER "PhysX Samples")
 
        ENDIF()
-ENDIF()
\ No newline at end of file
+ENDIF()

```

The problem with setting it to 3.12 is that Ubuntu 18.04 (LTS) only has cmake 3.10, is there a specific reason for wanting cmake 3.12 or what is just the one what was used when the CMakeLists.txt file was written?

AlesBorovicka, 2018-12-21:
Yes, cmake 3.12 is required as it fixes issues with latest xcode on mac. Lower version might be fine on linux though, but in general we require 3.12 because of xcode. On windows it might be 3.10-3.11 that is fine, there were some issues with vs2017...

salamanderrake, 2018-12-24:
As a side note, I just found out about this, so even on a system that doesn't install cmake >= 3.12 there is.
pip [CMake](https://pypi.org/project/cmake) page
So you can put that in your docs somewhere so people know they can install it.

For users local copy

> pip install --user cmake

For system wide install
>sudo pip install cmake

AlesBorovicka, 2018-12-26:
Thanks! I will add this to the documentation - the linux readme file.

AlesBorovicka, 2019-01-02:
Doc updated, changes will appear in next patch update

closed 2019-01-02 by AlesBorovicka