NVIDIAGameWorks/PhysX #13: License incompatibility of PhysX' BSD and FleX' Gameworks license

thebrandre, 2018-12-22:
To my understanding, we can't use PhysX-4.0 together with FleX because PhysX is now distributed under the 3-Clause BSD license which forbids the usage of the name NVIDIA without written consent.
This is the exact opposite of the [NVIDIA GAMEWORKS license](https://developer.nvidia.com/gameworks-sdk-eula) as it requires detailed attribution to NVIDIA in section 6.

Is there any recommendation on how to deal with that? I am quite certain you don't want to allow the usage of NVIDIA's name individually?
Or ... can you tell if there are any plans for FleX becoming available under BSD license?

gallonmate, 2018-12-22:
Flex SDK requires the "Nvidia Gameworks" logo in certain locations and specifying Nvidia Flex.

And furthermore if you make a commercial product with Flex SDK, you're supposed to contact Nvidia with details before releasing the product or videos.

"With Physx 4.0 Neither the name of NVIDIA CORPORATION nor the names of its contributors may be used to endorse or promote products derived from this software without specific prior written permission."

So the licenses are compatible.
Because you would have to contact nvidia if you wanted to release a commercial product with Flex SDK.
If it was a non-commercial product then you wouldn't need to give credit and you can't promote the name. 

Except with Unreal Engine 4, for that specific Flex integration, I could find no license or requirements regarding their name. Which is strange.

thebrandre, 2018-12-22:
Thanks for your quick answer. I can't follow your train of thought, though.

If I remember correctly, "Nvidia Gameworks" expects you to inform NVIDIA but not to wait on an answer/permission. So it got more restrictive by going BSD. This can't be as intended?

gallonmate, 2018-12-22:
Sorry I was actually going to delete my comment because it's something only nvidia can clarify and honestly their licensing is confusing. Especially now that I realize that UE4 gameworks doesn't have any license notifications or agreements that I could find, which brings up my own concerns about using ue4 gameworks branches.

But anyways it says "You are required to notify NVIDIA prior to use of the NVIDIA GameWorks SDK in the development of any commercial Application"

TimLeijten, 2018-12-23:
> Sorry I was actually going to delete my comment because it's something only nvidia can clarify and honestly their licensing is confusing. Especially now that I realize that UE4 gameworks doesn't [...]

If I'm correct, the UE4 FleX implementation is made by NVidia. So they probably have the rights for it.

0lento, 2018-12-25:
I think Nvidia staff has mentioned in past that what is in their UE4 fork goes under standard UE4 EULA. So you don't need to fulfill separate licensing conditions for those UE4 specific branches as long as you are fine with the terms you get from Epic.

JordiBravo, 2018-12-26:
> I think Nvidia staff has mentioned in past that what is in their UE4 fork goes under standard UE4 EULA. So you don't need to fulfill separate licensing conditions for those UE4 specific branches as long as you are fine with the terms you get from Epic.

Can anyone confirm this or find a source?

0lento, 2019-01-01:
> Can anyone confirm this or find a source?

https://forums.unrealengine.com/community/general-discussion/24447-nvidia-gameworks-integration?p=333178#post333178

0lento, 2019-01-01:
also https://forums.unrealengine.com/community/general-discussion/24447-nvidia-gameworks-integration?p=329199#post329199 (Mike works for Nvidia)

amoravanszky, 2019-01-08:
I tried to chase this up with our legal team but so far I did not get a conclusive answer.  My own view (IANAL and so on) is that this conflict is certainly not intentional and it seems obvious to me that us open sourcing PhysX under BSD is in no way intended to undermine the separate and independent licensing of other NVIDIA technologies under their own licensing terms.  So just go ahead and meet the attribution requirements of FLEX as needed, it will not cause you any trouble with the PhysX team or NVIDIA I promise.  You might not receive the right to use the NVIDIA branding through the PhysX BSD license, but you do get it through the FLEX license, and PhysX does not take it away.

amoravanszky, 2019-01-10:
Good news, I got the following guidance from our legal counsel, I believe this resolves the question:

_Clause 3 of the BSD is a restriction on how recipients of the software can use NV’s trademark.  So, for example, <company X> can’t pick up one of our BSD-licensed projects, include it in their product, and then advertise it as “Project X by NVIDIA!”  That’s it.  They get no TM usage rights FROM THIS LICENSE (other than the requirement to indicate the software’s origin are required by the license).  

If they get TM license rights (or obligations) from us elsewhere, as from a Flex license agreement with NV, then they can comply with those terms without conflicting with the license here._

thebrandre, 2019-01-11:
Thank you for the clarification! This resolves the question. :-)

closed 2019-02-28 by amoravanszky