NVIDIAGameWorks/PhysX #15: Project generation requires Python, how to avoid?

TimLeijten, 2018-12-23:
Hi

THis isn't really an issue, but it's something that really bothers me and would make it harder to use. You have decided to use python as part of the project generation, where I think CMake should easily be able to handle it all by itself. I am wondering, why include Python in the mix? And are you planning on removing it as a dependency?
The reason python bothers my is because I regularly build on platforms without python, and I regularly work with a git managed project, where I prefer full control over CMake, and because of python, I don't have that.

So, are you planning on removing Python as part of your build system? And is there a way to avoid it when I integrate it with my CMakeLists.txt?(E.g. what variables need to be set to where?)

Thanks!

AlesBorovicka, 2018-12-23:
Hi,
 no we do not plan to remove python dependency atm. We started to use just CMake and generate projects through batch files. But this approach was too restrictive and changing a cmake variable was not really possible within our development or on our builders (note that it is not possible for us to generate cmake projects with just cmake command, we need dependencies to be set). We had to use fixed settings, which is not very convenient. So we end up with creating presets that store our cmake switches settings and added a python script that parses the xml's and generates the final cmake command. 
If you wish to use directly cmake I recommend you to print out the final cmake command from the cmake_generate_projects.py script.
Regards,
Ales

TimLeijten, 2018-12-23:
> Hi,
> no we do not plan to remove python dependency atm. We started to use just CMake and generate projects through batch files. But this approach was too restrictive and changing a cmake variable was [...]

Thanks for your reply! I will look into the output generated by the cmake_generate_projects script. But I am wondering, if we are supposed to use the python script, then how do we integrate it into out CMake projects? Is there some tutorial on that?(I mean so that I can start using it in my game, instead of just compiling the binaries and samples)

Thanks!

AlesBorovicka, 2018-12-24:
You can include the cmake files into your project, but you need to set all variables required. So you need the variables from generate_projects.bat:
PhysX/physx/generate_projects.bat
Then you need to set all parameters that are generated in the python script, you can print them out and add them to your build. 
I am not sure you should do this anyway, there are lots of parameters just for PhysX compilation, is it a good idea to drag this into your system? External dependencies are usually build outside of projects. 

Anyway next year I can put some more information in the manual how to decouple the python, but note that it will never be easy, as PhysX does have dependencies itself and requires several cmake parameters to be set. Therefore the python script exist in the first place. So say for example if you want to use float precise math then you can change a bool in a preset and you are done.

TimLeijten, 2018-12-24:
> You can include the cmake files into your project, but you need to set all variables required. So you need the variables from generate_projects.bat:
> PhysX/physx/generate_projects.bat
> Then you need [...]

Hi, thanks for your reply. A tutorial on how to decouple from Python and set the Libraries myself would be very handy! 
I understand that one would normally build dll's/so's for something outside of the actual project, but my problem is is that I then have to manually add all the header's and linking files to my project. Which seems like a lot of work considering PhysX isn't particularly small.

So if I'm supposed to compile PhysX separately(which I can understand), then what is the recommended way to include the headers into my project?
And, what if I want to compile PhysX into my project?(Into the actual executable), would I be allowed to do so? And how would I do that?

Thanks!

AlesBorovicka, 2018-12-26:
You can use the install project, that will output the required include headers and libraries that you need for your project (note that the cudacontext headers are missing there now, need to fix that they are a 
part of different internal project).

But anyway as for the headers, all physx headers required are here:
physx\include
pxshared\include
Thats all you need as for the headers. Or am I missing something?

I dont understand you last question? When you link against the libraries that you require (most likely PhysX, PhysXCooking, PhysXCommon, PhysXFoundation) then your executable will compile. 

But again you can get the required cmake parameters out from the .bat and the final cmake command that gets parsed by the python script, that does replace 20+ possible batch files that would generate cmake projects for different platforms and platform targets.

Regards,
Ales

TimLeijten, 2018-12-26:
> You can use the install project, that will output the required include headers and libraries that you need for your project (note that the cudacontext headers are missing there now, need to fix that [...]

Ah thanks for the reply! I totally forgot about that. So it's basicly providing the include directories and providing which libraries need to be linked to? Ok, thanks!

I do want to comment on one thing, in your last paragraph you say not using the .bat file/python would require a lot more files. But I have to disagree, CMake allows to detect what platform you are using, and can adjust for that. It can also do it on a per compiler or architecture base. That is kinda what CMake was made for. Now I'm not trying to be mean or saying you didn't do it right, I'm just saying that I think it is unecesery and it should be not that hard with just CMake files.

But anyways, thanks for your help and have a nice day!

closed 2019-02-28 by amoravanszky

n8vm, 2019-04-20:
IMO, this issue should still be opened.

“External dependencies are usually build outside of projects.”

This is a bold assumption. Maybe this is true for some Physx users, but not for many open source projects using git submodules (See Matt Phar’s PBRT as an example. I do something similar for my game engine, and I know many others who run a similar workflow). If I cannot easily build Physx within my own build system, I cannot guarantee Physx will work on all platforms my engine builds on. As it stands, it’s difficult if not impossible to hack in Physx’s custom python build system into my own CMake build system, and I cannot afford to commit many different sets of large prebuilt binary files to a repository either.

Competing solutions like Bullet 3 allow submoduling and support pure CMake build systems. If I want to include Bullet in my project, submoduling and adding bullet using the CMake “add_subdirectory” is trivial. CMake options are not hard to manage either.

It sounds like, due to some internal technicalities, Physx’s CMake build system has been unnecessarily over engineered. This complexity makes it impossible for me and many others to include Physx reliably in their projects. 

What exactly makes a pure CMake build system so infeasible to PhysX, but so natural for Bullet physics?

AlesBorovicka, 2019-04-23:
Hi,
we tried to use more directly CMake, but we run into issues with that.
 1. there were simply too many different configuration scripts to generate projects, not only for each platform (note that PhysX does support also consoles like xb1, ps4, switch, which cannot be included into public - hence the per platform cmake directories). Additionally even for different customers we have to support different projects settings even for windows for example static/dynamic libraries switches.
 2. we could have enforce people to use just stock cmake and fill in the required parameters, but that is pretty hard, as we have some binary dependencies that come as packages and those paths/variables have to be set. Note that using stock cmake is also a problem, as cmake does not support all platforms out of the box. 
 3. so we moved all the possible switches to an external xml files which can be configured per platform/project. This works very well for us, as we can have one python file that puts together the cmake command, you can still retrieve the cmake command from the python script though.