NVIDIAGameWorks/PhysX #18: CharacterKinematic static library issues on Windows

al2950, 2018-12-24:
Hi

The CharacterKinematic is built by default as a static library, but it looks like it can be configured as a dynamic library. When building everything is fine, however when including in a project, in windows, it causes linker errors. It is caused by this code in PxCharacter.h;

```
// define API function declaration
#if defined PX_PHYSX_STATIC_LIB || defined PX_PHYSX_CHARACTER_STATIC_LIB
	#define PX_PHYSX_CHARACTER_API
#else
	#if PX_WINDOWS
		#if defined PX_PHYSX_CHARACTER_EXPORTS
			#define PX_PHYSX_CHARACTER_API __declspec(dllexport)
		#else
			#define PX_PHYSX_CHARACTER_API __declspec(dllimport)
		#endif
	#elif PX_UNIX_FAMILY
		#define PX_PHYSX_CHARACTER_API PX_UNIX_EXPORT
    #else
		#define PX_PHYSX_CHARACTER_API
    #endif
#endif
```

The issue is a static library is 'exported' with no special declarations but if using the the library on windows with no knowledge, it uses __declspec(dllimport), which results in a linker error. This can be easily solved by the end user ensuring PX_PHYSX_CHARACTER_STATIC_LIB is defined. However its not very obvious. May I suggest either adding some documentation around this, or even better get CMAKE to create a BuildSettings.h type file which contains the pre-processor defines that are decide at CMAKE configure time. ie PX_PHYSX_CHARACTER_STATIC_LIB 

Thanks :)

PS. Apart from this and the [other issue](https://github.com/NVIDIAGameWorks/PhysX/issues/17) i found, the new build system works great

AlesBorovicka, 2018-12-26:
Hi,
 right, we did not shipped anymore this library as dll's but it is true, that with cmake you can configure it to build it as a dll. 
You would need the PX_PHYSX_CHARACTER_EXPORTS defined in first place in the cmake files for the character kinematics dll. There are unfortunately a bit more things to do to have this library a dll for us. We would have to also add a delay load hooks in there... so already since 3.4 we kind of not officially support this library as a dll. But yes with cmake you should be able to configure it so, but I think the only thing missing actually is to add PX_PHYSX_CHARACTER_EXPORTS instead of PX_PHYSX_CHARACTER_STATIC_LIB in the PhysXCharacterKinematic.cmake file. Let me try that we could add it, but we wont probably add the hooks above. 
Regards,
Ales

AlesBorovicka, 2018-12-27:
Ah I misunderstood you, yes that might be the best to remove the import/export stuff. We will discuss this internally.

aiplay, 2019-01-04:
Same problem on me, until I set PX_PHYSX_CHARACTER_STATIC_LIB macro on my project.

AlesBorovicka, 2019-01-08:
The import/export macros have been removed, change will be in next update.

closed 2019-01-08 by AlesBorovicka

al2950, 2019-01-08:
Great, thank you. 

Out of interest is there a set release cycle or will you publish updates when you feel its relevant?

AlesBorovicka, 2019-01-08:
No there will be no set release cycle for sure. We mostly gather some fixes and then after some smoke tests try to update. 
So I expect we will make update most likely next week.