NVIDIAGameWorks/PhysX #24: Arbitrary axis for PxArticulationJointReducedCoordinate joints

christoph-jaehne, 2019-01-07:
Hi,
I'm working with reduced coordinate articulations for the purpose of simulating robots in generalized coordinates. As far as I understand from the code and the docs, when creating an articulation joint `PxArticulationJoint` I can only configure one of the discrete values defined in `PxArticulationAxis` like `eTWIST`, `eSWING1`, `eSWING2`, `eX`, `eY`, `eZ`an `eCOUNT` as axis of motion. Are there any plans to support arbitrary unit vectors as axis of motion for revolute and prismatic joints at some point?
This seems necessary to comply with structures as defined by generic URDFs..

kstorey-nvidia, 2019-01-07:
You have access to the joint frames through setParentPose and setChildPose methods. You can compute a rotation that will align the twist (1,0,0), swing1(0,1,0) and swing2(0,0,1) axes to the desired set of orthogonal axes.

We don't plan to support arbitrary axes because this introduces the problematic case of users defining axes that are not orthogonal to each-other. These configurations would not simulate correctly. By restricting the axes to the 3 principal axes with the ability to orient the frame, this avoids the risk of users misconfiguring the joints.

christoph-jaehne, 2019-01-08:
The problem with that solution is that if i correct/align the joint pose on parent side such that one axis is aligned (e.g. with setParentPose) I cannot revert that change on the child side with setChildPose, as the correcting transformation depends on the joints degree of freedom and is therefore not static. Ultimately that means that I permanently change the child frame, and therefore have to adapt all origins and poses that are defined in that frame like inertias, collision spaces and consecutive joints.

Would it be an option to check provided axes for orthogonality and reject them if applicable to ensure orthogonal axes?

kstorey-nvidia, 2019-01-08:
A colleague of mine is porting a urdf loader to a snippet to demonstrate how to do this. Internally, we are using 3rd party xml loaders, so we need to verify licensing to confirm whether we can distribute these along with source to the loader and which urdf file we can distribute as a demonstration (again, I am unsure of which licensing terms are involved so we would need to confirm to avoid any risk of breaching licensing terms). Once it's ready, this will be included in our next code drop after some internal testing and reviews. However, as this might not be available for a little while, I'll try to unblock you.

Could you provide an example of something that you think doesn't work? We have loaded many robot models with PhysX and have not found configurations it couldn't represent. AFAIK, most urdf models have only a single dof per-joint (in fact, I believe this is all many engines support). However, we have also found no issues loading various mjcf files that contain joints with 2 or 3 degrees of freedom. If you could provide a particular case that you are struggling with, we might be able to help solve the issue.

There shouldn't be any need to adjust properties like inertias, collision spaces or for any joint frame adjustments to propagate to consecutive joints if you rotate the joint frame. Collisions and inertias are defined in link space - you define this space when you create a link and pass the pose to it. The joint space (used to define degrees of freedom) is a separate concept that defines the anchor location of the joint and its alignment in relation to the parent and child's link space.

The only requirement should be to calculate a rotation that will rotate the desired axis (e.g. eTWIST(1,0,0)) to the arbitrary axis you require to align with (origin). If you have just 1 dof, the following snippet should do that. Please note that I adapted from our urdf loader to use PhysX maths types, (that code uses its own internal maths library) so apologies for any typos I introduced:

PxTransform origin = j->origin;

PxQuat alignQ = PxShortestRotation(PxVec3(1.f, 0.f, 0.f), origin);

origin.q = j->origin.q * alignQ;

Please note that this is operating on the joint origin, not the link origin. In this example, we would have already created and configured the parent and child links and any collision shapes being used from the values extracted from the urdf file. 

In the loader, origin is applied as follows:

PxTransform ptran = parentLink->getGlobalPose();
PxTransform ctran = childLink->getGlobalPose();
Transform ppose = ptran.getInverse() * origin;
Transform cpose = ctran.getInverse() * origin;

physx::PxArticulationJointReducedCoordinate* joint = static_cast<physx::PxArticulationJointReducedCoordinate*>(childLink->getInboundJoint());	
joint->setParentPose(ppose);
joint->setChildPose(cpose);

christoph-jaehne, 2019-01-10:
Thanks for this helpful and quick answer. I think I see now what the missing piece of information on my side was and maybe you can confirm or correct me on the following points:
1. When adding a new link to an articulation with `articulation->createLink(parent, link_pose)` the link_pose is NOT relative to the parent frame but in global/world coordinates.
2. The line `PxTransform origin = j->origin;` also gives the joint origin in global/world coordinates.
If these 2 points hold I'm happy and understand the solution ;)

kstorey-nvidia, 2019-01-15:
This is correct.

The link pose provided to articulation->createLink should be provided in world space. 

The joint origin in the urdf file is in model space. Similarly, the origin for joint links are defined in the same model space. 

It's quite common that you might want to apply an additional transform to these to model space transform the links/joints into world space (e.g. if you want to place the robot somewhere in your world). The snippet I provided assumes that the robot was instantiated in a world space that coincided  with model space.

christoph-jaehne, 2019-01-16:
Thanks for the help.

erwincoumans, 2019-01-18:
I spend a few days integrating PhysX 4.0  into [PyBullet](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3), re-using our [URDF Parser](https://github.com/bulletphysics/bullet3/blob/master/examples/Importers/ImportURDFDemo/UrdfParser.cpp) (re-implemented based on the original ROS URDF loader, and using tinyxml2). Indeed, you can compute a relative transform between reference (1,0,0) and the desired joint axis.

This is a snippet of the conversion:
[code]
		btVector3 refAxis(1, 0, 0);
		btVector3 axis = jointAxisInJointSpace;
		btQuaternion axis2ReferenceRot = shortestArcQuat(refAxis, btVector3(axis[0], axis[1], axis[2]));
[/code]

closed 2019-02-28 by amoravanszky