NVIDIAGameWorks/PhysX #26: Why do single-sided objects and double-sided objects take the same time when performing scene queries?

Qinja, 2019-01-08:
I am performing a raycast on a scene. Regardless of whether my scene is a single-sided model or a two-sided model, the speed of raycast is the same? It is reasonable to say that the higher the number of faces, the greater the complexity and the more time it takes.
In addition, do I need to set one-sided or two-sided when raycast? I looked at the documentation and found that the scene query has no options for  " both sides " but the geometry query has.

Thanks. : )

PierreTerdiman, 2019-01-08:
> Regardless of whether my scene is a single-sided model or a two-sided model, the speed of raycast is the same?

It should be pretty much the same speed, unless your rays go through a lot of back faces before reaching a front face. But that shouldn't happen very often with most meshes. So basically, indeed, single-sided or two-sided shouldn't make much of a difference w.r.t. raycast performance.

> It is reasonable to say that the higher the number of faces, the greater the complexity and the more time it takes.

It is "reasonable" but there is a lot more than this to raycast performance. If you go into the details, it ultimately depends not just on the number of faces but also their sizes and spatial organization, how much they overlap each other, whether the triangles are uniformly distributed or not, and a hundred other details.

> do I need to set one-sided or two-sided when raycast?

It depends what you want to achieve.

> the scene query has no options for " both sides "

What about PxHitFlag::eMESH_BOTH_SIDES?

Qinja, 2019-01-08:
I get it .Thanks.

closed 2019-01-08 by Qinja