NVIDIAGameWorks/PhysX #28: Compile ios64, the static lib's size is too large

aiplay, 2019-01-09:
After I compiled PhysX SDK with xcode, the libPhysX_static_64.a is 221Mb.
When set the Generate Debuy Symbols is False, the generate static lib is 186Mb, but still too large for our git project.
So, how can I to reduce the size of libPhysX_static_64.a ??

AlesBorovicka, 2019-01-09:
You can compile only architecture that you need. It does contain now three architectures, you can select to compile only target architecture, so the size will be roughly the third.

aiplay, 2019-01-09:
Only use arm64 architechture, the  libPhysX_static_64.a  size is 61Mb, is it normal ?

AlesBorovicka, 2019-01-09:
Without the debug symbols it could be

closed 2019-01-28 by AlesBorovicka

zhongfugao, 2022-04-19:
I met this issue as well, any updates on it?