NVIDIAGameWorks/PhysX #31: PX_PHYSX_STATIC_LIB should apply to foundation as well

jamesdolan, 2019-01-12:
PX_PHYSX_STATIC_LIB controls the dllimport/dllexport of most of the libraries except for foundation.

This means that to include headers from static lib builds I need to do...
```
#define PX_PHYSX_STATIC_LIB
#define PX_FOUNDATION_DLL 0
```
...which seems a bit clunky. Ideally a simple config.h could be generated that gets included in both the physx build and the application build to ensure consistent environment. Or at the very least reduce it to a single define (e.g. PX_PHYSX_STATIC_LIB).

AlesBorovicka, 2019-01-13:
Hi James,
 You are right, since foundation is now a part of the PhysX SDK, there should be just one define. 
Regarding the PxConfig.h, I was considering that, it was already suggested in another post. I think we can generate it, sounds like a good thing to do.

Ales

AlesBorovicka, 2019-01-28:
Should be fixed, foundation does use PX_PHYSX_STATIC_LIB too, additionally PxConfig.h header file is generated, that can be used to avoid the PX_PHYSX_STATIC_LIB.

closed 2019-01-28 by AlesBorovicka