NVIDIAGameWorks/PhysX #33: Can we get rid of the nested header search paths?

jamesdolan, 2019-01-12:
I realize supporting other build systems isn't something that physx would want to officially do, but I do think some more rigid conventions would make it easier for those that have to do it ourselves for one reason or another.

Integrating PhysX into a our build system required a lot more verbosity than I think really should be necessary. The convention that I typically see in projects that have "include" and "src" directories is that "include" is the sole search path required for dependencies to use said library.

To make PhysX itself build I had to add about **60** search paths just for physx to find other parts of itself. The vast majority of those search paths were under other search paths.

For example I had to not only add `physx/include`, but also `physx/include/common`, `physx/include/geometry`, and so forth. Which means its possible to include the same file with different paths (`<PxPvd.h>` and `<pvd/PxPvd.h>`).

There are also internal libraries that require headers that are in other libraries "src" directories, which seems like mixing two conventions. See the `PX_COOKING` issue below for what I think is an example of the fallout of this.

Another minor issue is there are some platform specific files stashed under "unix" directories and some under "linux", which means I can't just expand a platform name variable to build those files, I need to do some project-specific branching.

And one last thing, without `PX_COOKING` defined I was getting `#error Do not include anymore!`. With a message like that it sounds like something that should either just be removed from the repository, or the error just removed. If I had to hazard a guess, its because some other non-cooking library was including `physxcooking/src/MeshCleaner.h` and to ensure that didn't happen anymore someone made including that file outside cooking fail?

erwincoumans, 2019-01-13:
Thanks James! Can you share list of those search paths and all the preprocessor definitions?

jamesdolan, 2019-01-13:
Sure, stashed them in python variables, assumes you can glob them yourself though...
```
  SourceFiles  = [
    'physx/source/common/src/*.cpp',
    'physx/source/common/src/{PhysXPlatform}/*.cpp',
    'physx/source/fastxml/src/*.cpp',
    'physx/source/foundation/src/*.cpp',
    'physx/source/foundation/src/{PhysXPlatform}/*.cpp',
    'physx/source/geomutils/src/*.cpp',
    'physx/source/geomutils/src/*/*.cpp',
    'physx/source/immediatemode/src/*.cpp',
    'physx/source/lowlevel/api/src/*.cpp',
    'physx/source/lowlevel/common/src/*/*.cpp',
    'physx/source/lowlevel/software/src/*.cpp',
    'physx/source/lowlevelaabb/src/*.cpp',
    'physx/source/lowleveldynamics/src/*.cpp',
    'physx/source/physx/src/*.cpp',
    'physx/source/physx/src/buffering/*.cpp',
    'physx/source/physx/src/device/{PhysXPlatform}/*.cpp',
    'physx/source/physx/src/gpu/*.cpp',
    'physx/source/physxcharacterkinematic/src/*.cpp',
    'physx/source/physxcooking/src/*.cpp',
    'physx/source/physxcooking/src/convex/*.cpp',
    'physx/source/physxcooking/src/mesh/*.cpp',
    'physx/source/physxextensions/src/*.cpp',
    'physx/source/physxextensions/src/*/*.cpp',
    'physx/source/physxextensions/src/*/*/*.cpp',
    'physx/source/physxmetadata/core/src/*.cpp',
    'physx/source/physxmetadata/extensions/src/*.cpp',
    'physx/source/physxvehicle/src/*.cpp',
    'physx/source/physxvehicle/src/physxmetadata/src/*.cpp',
    'physx/source/pvd/src/*.cpp',
    'physx/source/scenequery/src/*.cpp',
    'physx/source/simulationcontroller/src/*.cpp',
    'physx/source/task/src/*.cpp',
  ]
  PublicHeaderSearchPaths = [
    'physx/include',
    'physx/include/characterkinematic',
    'physx/include/common',
    'physx/include/cooking',
    'physx/include/extensions',
    'physx/include/geometry',
    'physx/include/geomutils',
    'physx/include/vehicle',
    'pxshared/include',
  ]
  PrivateHeaderSearchPaths = [
    'physx/source/common/include',
    'physx/source/common/src',
    'physx/source/fastxml/include',
    'physx/source/filebuf/include',
    'physx/source/foundation/include',
    'physx/source/geomutils/include',
    'physx/source/geomutils/src',
    'physx/source/geomutils/src/ccd',
    'physx/source/geomutils/src/common',
    'physx/source/geomutils/src/contact',
    'physx/source/geomutils/src/convex',
    'physx/source/geomutils/src/distance',
    'physx/source/geomutils/src/gjk',
    'physx/source/geomutils/src/hf',
    'physx/source/geomutils/src/intersection',
    'physx/source/geomutils/src/mesh',
    'physx/source/geomutils/src/pcm',
    'physx/source/geomutils/src/sweep',
    'physx/source/lowlevel/api/include',
    'physx/source/lowlevel/common/include',
    'physx/source/lowlevel/common/include/collision',
    'physx/source/lowlevel/common/include/pipeline',
    'physx/source/lowlevel/common/include/utils',
    'physx/source/lowlevel/software/include',
    'physx/source/lowlevelaabb/include',
    'physx/source/lowleveldynamics/include',
    'physx/source/physx/src',
    'physx/source/physx/src/buffering',
    'physx/source/physx/src/device',
    'physx/source/physxcooking/src',
    'physx/source/physxcooking/src/convex',
    'physx/source/physxcooking/src/mesh',
    'physx/source/physxextensions/src',
    'physx/source/physxextensions/src/serialization/Binary',
    'physx/source/physxextensions/src/serialization/File',
    'physx/source/physxextensions/src/serialization/Xml',
    'physx/source/physxmetadata/core/include',
    'physx/source/physxmetadata/extensions/include',
    'physx/source/physxvehicle/src',
    'physx/source/physxvehicle/src/physxmetadata/include',
    'physx/source/pvd/include',
    'physx/source/scenequery/include',
    'physx/source/simulationcontroller/include',
    'physx/source/simulationcontroller/src',
  ]
```

erwincoumans, 2019-01-13:
Great, thanks!
Do you mind sharing all the preprocessor definitions?

jamesdolan, 2019-01-13:
Sure, although luckily its a shorter list (wish it was just `PX_PHYSX_STATIC_LIB` though)...
```
  CPPDefines = [
    'PX_PHYSX_STATIC_LIB',
    'PX_FOUNDATION_DLL=0',
    'PX_COOKING', # needed for "#error Do not include anymore!"
  ]
```

sschirm, 2019-01-21:
Thanks James, 
I think many of you complaints are very legit, and I hope we can put some effort into making this better!

jamesdolan, 2019-01-21:
Don't think of it as complaints, but rather love in the form of feedback :)

Just trying to help you, umm... Make PhysX Great Again.