NVIDIAGameWorks/PhysX #35: Updating vehicles wheels and drive simulation data after setup

siliace, 2019-01-14:
Hello,

I want to implement a system of vehicle destruction during the simulation. This system would change things like the maximum torque/speed of the engine, the ackermaan geometry, suspensions...

I wonder what is the more suitable way to implement this kind of thing with the PhysX vehicle SDK? 

Is it possible to do something like :
`void Car::UpdatePhysXData(const PxVehicleWheelsSimData& newSimData, const PxVehicleDriveSimData4W& newDriveSimData)
{
drive->setup(&PxGetPhysics(), 
                     mDynActor, 
                     newSimData,
                     newDriveSimData, 
                     mNbWheels - 4);
}`
My other concern about this solution is performance wise, can it have a significant impact on it?
Or should I emulate things like a damaged steer/engine by editing the `PxVehicleDrive4WRawInputData` I use to update my vehicle? This would be easier to do but would be more limiting.

Thanks by advance,

Siliace

gyeomannvidia, 2019-01-14:
Hi,

The cost of changing vehicle data at run-time should be quite cheap.  A few examples might point to a solution to your particular problem.

Let's say you want to change the longitudinal stiffness of tire 0 at run-time.  It might look like this:

```
PxVehicleTireData tireData = myVehicle.mWheelsSimData.getTireData(0);
tireData.mLongitudinalStiffnessPerUnitGravity = newValue;
myVehicle.mWheelsSimData.setTireData(0, tireData);
```

Another example might be changing the peak engine torque:

```
PxVehicleEngineData  engineData = myVehicle.mDriveSimdata.getEngineData();
engineData.mPeakTorque = newValue;
myVehicle.mDriveSimData.setEngineData(engineData);
```

You can repeat this pattern for almost any parameter.

The only caveat is that the data is not changed while PxVehicleUpdates is executing. 

Cheers,

Gordon

siliace, 2019-01-14:
You saved my day, thank you very much !

closed 2019-01-14 by siliace