NVIDIAGameWorks/PhysX #36: Linking erros PxDefaultErrorCallback

jpszerzp, 2019-01-14:
Hi,

I was trying to build the physx solution when I got following problems:

1. Building solutions with either debug or checked will generate include errors for d3dx9.h and d3d11tex.h like this:
`2>c:\physx-4.0\physx\samples\sampleframework\renderer\src\d3d9\d3d9renderer.h(42): fatal error C1083: Cannot open include file: 'd3dx9.h': No such file or directory (compiling source file C:\PhysX-4.0\physx\samples\sampleframework\renderer\src\d3d9\D3D9RendererVertexBuffer.cpp)
2>c:\physx-4.0\physx\samples\sampleframework\renderer\src\d3d11\d3d11rendererutils.h(38): fatal error C1083: Cannot open include file: 'd3dx11tex.h': No such file or directory (compiling source file C:\PhysX-4.0\physx\samples\sampleframework\renderer\src\d3d11\D3D11RendererMaterial.cpp)`

2. The above errors do not prevent the libs from generating though. I can find PhysXCommon_64.lib, PhysX_64.lib, PhysXFoundation_64.lib and PhysXCooking_64.lib and their dlls under the path physx\bin\win.x86_64.vc141.mt\checked (or \debug) as I building with x64 VS 2017. So here is what I did:

- In VC++ Directories -> Include Directories, I included physx\include and pxshared\include
- In VC++ Directories -> Library Directories, I included local\thirdparty\physx, which is a local directory I used to put the 4 lib files above
- In Linker->General->Additional Library Directories, I included local\thirdparty\physx again
- In Linker->Input, I included names of the 4 libs above, in order, separated with ';'
- I then copied the 5 dlls (plus PhysXGpu_64.dll) to the same folder as my application's .exe
- In my TheApp.hpp (my startup class) I included the lines
`#pragma comment(lib, "PhysX_64.lib")
#pragma comment(lib, "PhysXCommon_64.lib")
#pragma comment(lib, "PhysXCooking_64.lib")
#pragma comment(lib, "PhysXFoundation_64.lib")`

But then when I tried to create
`PxDefaultErrorCallback m_DefaultErrorCallback;`
as a member of TheApp, the linking error happens
`1>TheApp.obj : error LNK2019: unresolved external symbol "public: __cdecl physx::PxDefaultErrorCallback::PxDefaultErrorCallback(void)" (??0PxDefaultErrorCallback@physx@@QEAA@XZ) referenced in function "public: __cdecl TheApp::TheApp(void)" (??0TheApp@@QEAA@XZ)
1>TheApp.obj : error LNK2019: unresolved external symbol "public: virtual __cdecl physx::PxDefaultErrorCallback::~PxDefaultErrorCallback(void)" (??1PxDefaultErrorCallback@physx@@UEAA@XZ) referenced in function "int `public: __cdecl TheApp::TheApp(void)'::`1'::dtor$0" (?dtor$0@?0???0TheApp@@QEAA@XZ@4HA)`

I am not sure what is going wrong with my linking. Can someone share some insights? Thx.

John

AlesBorovicka, 2019-01-14:
Hi,
 you are missing dxsdk. It is mentioned as requirement in the windows readme and you should see a warning during the generate_projects.bat. 
Regards,
Ales

kstorey-nvidia, 2019-01-16:
Regarding the linker error in your application, did you link against the PhysX extensions library? This is where the default error callback, default CPU dispatcher etc. are provided. Most game engines will provide their own implementations of these, integrated into their own internal engine infrastructure, so these features are provided as a reference implementation in an optional statically-linked library.

closed 2019-01-28 by AlesBorovicka

kmiloarguello, 2020-06-02:
Hi, I would like to reopen this issue due to I am having the same problem whenever I try to run PhysX. (image) I am able to execute the code snippets inside of physx folder, however if i want to import the library into my own project I got those errors.

[image]

I verified whether I have installed everything, also DirectX SDK (image2), and everything is already installed.

[image]

I don't know what I missing, and I will appreciate any help solving this issue.

Sorry for the inconvenience.

Camillo.

kstorey-nvidia, 2020-06-02:
As Ales mentioned in a previous reply, you are getting linker errors referencing things in the PhysX extensions library. Please verify that you are linking your application against PhysX extensions.

kmiloarguello, 2020-06-02:
@kstorey-nvidia Thank you for your answer.

Yes, in fact my error is a "linker error", however I've been reading the documentation and the readme for windows, and the docs aren't clear in the way to reference the library files. As I understood, I copied those files (image) from for example win.x86_64.vc142.mt, from the debug folder.

[image]

Then I paste them into my debug folder lib. And I called into the Aditional Dependencies properties.

[image]

My question is, if I chose the checked build, do I really need the debug files .lib? Because if I don't call those files, I will get the same PxDefaultErrorCallback. 

I really appreciate your help.

Camillo.

kstorey-nvidia, 2020-06-02:
You should use the .lib files built by the build config you are using. If you are using debug PhysX DLLs, you should use the debug PhysX static libraries. If you are using checked, you should use checked PhysX static libraries and so on for Profile or Release.

kmiloarguello, 2020-06-02:
Thank you so much. It works correctly.

pouleyKetchoupp, 2020-12-09:
For the error on missing `'d3dx9.h'`, I've installed dxsdk, re-generated projects and got no warning but the error was still present.

If anybody is struggling on the same thing, you can manually add `$(DXSDK_DIR)Include;` to project includes in `Configuration Properties > VC++ Directories > Include Directories`.