NVIDIAGameWorks/PhysX #38: "Illegal BroadPhaseUpdateData" with eSIMULATION_SHAPE and eENABLE_SPECULATIVE_CCD

ikkah, 2019-01-15:
Hi,

I think I've found a bug that probably has started occurring at some point after "PhysX 3.4.1, APEX 1.4.1 Release @2284554". I can reproduce the bug with latest PhysX 3.4, but I haven't tried to reproduce it on PhysX 4.0.

Here's a quick repro for it:

>physx::PxSceneDesc sceneDesc( PxGetPhysics().getTolerancesScale() );
sceneDesc.cpuDispatcher = physx::PxDefaultCpuDispatcherCreate( 1 );
sceneDesc.filterShader = physx::PxDefaultSimulationFilterShader;
auto scene = PxGetPhysics().createScene( sceneDesc );
>
>physx::PxTransform transform( physx::PxVec3( 0.0f, 0.0f, 0.0f ), physx::PxQuat( 0.0f, physx::PxVec3( 0.0f, 0.0f, 1.0f ) ) );
auto actor = PxGetPhysics().createRigidDynamic( transform );
actor->setRigidBodyFlag( physx::PxRigidBodyFlag::eENABLE_SPECULATIVE_CCD, true );
>
>auto material = PxGetPhysics().createMaterial( 0.0f, 0.0f, 0.0f );
physx::PxBoxGeometry geometry( 0.5f, 0.5f, 0.5f );
physx::PxShape* shape = scene->getPhysics().createShape( geometry, *material );
shape->setFlag( physx::PxShapeFlag::eSIMULATION_SHAPE, false );
actor->attachShape( *shape );
>
>scene->addActor( *actor );
scene->simulate( 1 / 60.0f );
scene->fetchResults( true );

PhysX emits an "Illegal BroadPhaseUpdateData" error and it also triggers an assert PX_ASSERT(mGroups[handle] != Bp::FilterGroup::eINVALID); in SimpleAABBManager::updateAABBsAndBP.

kstorey-nvidia, 2019-01-15:
Thanks for reporting the issue and providing such easy repro instructions. I've reproduced and fixed the issue. It's related to enabling speculative CCD and disabling simulation on the shape.

The fix is to change this code:

while ((shapeSim = iterator.getNext()) != NULL)
{
	changedMap.growAndSet(shapeSim->getElementID());
}

in Sc::Scene::preRigidBodyNarrowPhase(PxBaseTask* continuation)

to this:

while ((shapeSim = iterator.getNext()) != NULL)
{
	if(shapeSim->getFlags() & PxShapeFlag::eSIMULATION_SHAPE)
		changedMap.growAndSet(shapeSim->getElementID());
}

I'll check in fixes for these to 3.4 and 4.0 (I didn't confirm yet but I assume that 4.0 probably has the same issue).

This fix will be pushed to GitHub in the next release, which should come soon. In the meantime, feel free to modify the code as described above to unblock you.

ikkah, 2019-01-16:
Awesome, thanks for the extremely quick response! I applied your code change and the issue seems to be fixed.

closed 2019-01-16 by ikkah