NVIDIAGameWorks/PhysX #39: CMake generate_projects.sh problem on Arch Linux

datgame, 2019-01-16:
CMake 3.13.2
```PhysX/physx% bash generate_projects.sh
Preset parameter required, available presets:
(0) mac64 <--- macOS Xcode PhysX general settings
(1) ios64 <--- iOS Xcode PhysX general settings
(2) linux-aarch64 <--- Linux-aarch64 gcc PhysX SDK general settings
(3) linux <--- Linux clang PhysX SDK general settings
Enter preset number: 3
Running generate_projects.bat linux
Using preset xml: buildtools/presets/public/linux.xml
Target platform: linux using compiler: clang
PM_CMakeModules_PATH: /files/thirdparty/PhysX/physx/../externals/cmakemodules
PM_PATHS: /files/thirdparty/PhysX/physx/../externals/opengl-linux
Cmake: cmake
Not searching for unused variables given on the command line.
PHYSX ROOT /files/thirdparty/PhysX/physx
PhysX Build Platform: linux
Using CXX Compiler: /usr/bin/clang++
CMake Error at /files/thirdparty/PhysX/externals/cmakemodules/GetCompilerAndPlatform.cmake:244 (IF):
  if given arguments:

    "STREQUAL" "x86_64-unknown-linux-gnu" "OR" "STREQUAL" "x86_64-linux-gnu"

  Unknown arguments specified
Call Stack (most recent call first):
  /files/thirdparty/PhysX/externals/cmakemodules/NvidiaBuildOptions.cmake:99 (GetPlatformBinName)
  /files/thirdparty/PhysX/physx/source/compiler/cmake/CMakeLists.txt:73 (INCLUDE)

-- Configuring incomplete, errors occurred!
See also "/files/thirdparty/PhysX/physx/compiler/linux-debug/CMakeFiles/CMakeOutput.log".```

etc

AlesBorovicka, 2019-01-16:
Hmm looks like that CMAKE_LIBRARY_ARCHITECTURE is not set in your case.
You can try to add the CMAKE_LIBRARY_ARCHITECTURE and set it to "x86_64-linux-gnu". But you might need to set more stuff. The custom linux toolchain files are in externals/cmakemodules/linux, maybe you need a toolchain file on your system.

sapphous, 2019-01-25:
I'm not sure what all is necessary to properly fix this. But it seems to be working after the following hack:

After line 234 of physx/buildtools/cmake_generate_projects.py, add `outString = outString + ' -DCMAKE_TOOLCHAIN_FILE=' + os.environ['PM_CMakeModules_PATH'] + '/linux/linuxRegular.cmake'`. Then add a file externals/cmakemodules/linux/LinuxRegular.cmake containing `SET(CMAKE_CROSSCOMPILING FALSE)
SET(CMAKE_SYSTEM_NAME Linux)
SET(CMAKE_SYSTEM_VERSION 1)

SET(CMAKE_SYSROOT ${LINUX_ROOT})
SET(CMAKE_LIBRARY_ARCHITECTURE x86_64-linux-gnu)`.

datgame, 2019-01-26:
Thanks, @sapphous !
That worked. Just had to fix the casing in your paste of "LinuxRegular"
After line 234 of physx/buildtools/cmake_generate_projects.py, add 
`outString = outString + ' -DCMAKE_TOOLCHAIN_FILE=' + os.environ['PM_CMakeModules_PATH'] + '/linux/LinuxRegular.cmake'`
Then add a file externals/cmakemodules/linux/LinuxRegular.cmake containing ```SET(CMAKE_CROSSCOMPILING FALSE)
SET(CMAKE_SYSTEM_NAME Linux)
SET(CMAKE_SYSTEM_VERSION 1)
SET(CMAKE_SYSROOT ${LINUX_ROOT})
SET(CMAKE_LIBRARY_ARCHITECTURE x86_64-linux-gnu)```

datgame, 2019-10-14:
got latest from git today (4.1) in a clean new folder and the same issue is still there. 
the same fix works still though.

phcerdan, 2019-11-04:
Facing this as well. No need to add the LinuxRegular.cmake file, just add one line in https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/buildtools/cmake_generate_projects.py#L263 
```diff
 elif self.compiler == 'clang':
+ outString = outString + ' -DCMAKE_LIBRARY_ARCHITECTURE=x86_64-linux-gnu'

```

phcerdan, 2019-11-06:
After some research, using CMAKE_LIBRARY_ARCHITECTURE to detect the platform is wrong. That variable is engineered to work with Debian-derived distros, but not in general (i.e, not in RHEL, Arch, etc), see [bug report mozilla](https://bugzilla.redhat.com/show_bug.cgi?id=1531678) and [CMake docs](https://cmake.org/cmake/help/latest/variable/CMAKE_LANG_LIBRARY_ARCHITECTURE.html).

The solution is to use `CMAKE_CXX_COMPILER_ID` if you want to distinguish between clang, and arm. As you did for `windows` in the same CMake file.

Will open a PR shortly.