NVIDIAGameWorks/PhysX #40: Consider using LFS to store binaries

snichols, 2019-01-16:
There's quite a few binary files in this repository... prebuilt executables, libraries, textures, etc.  I'd imagine the history for those files will get quite large as this repo moves forward.  Why not use `git-lfs` to store those files?

jamesdolan, 2019-01-18:
I would caution against LFS, its extra steps git users need to perform, its not straightforward to install on all environments, and the builtin github download-as-zip feature won't work as expected.

I would though strongly suggest moving the kaplademo to a separate repository. Don't ship pre-compiled libs for the external dependencies (glew for instance is super easy to build, why ship a binary that needs to be maintained?). At that point it should really just be the PhysXGPU lib.

nvidia-bgaldrikian, 2019-01-18:
We already deliver the SDK packaged through Packman on a public server, as well as other libraries and resources.  I don’t see a reason we couldn’t package up demos as well and provide a batch file to pull them.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------