NVIDIAGameWorks/PhysX #41: Crash in ProcessAggPairsParallelTask

ikkah, 2019-01-17:
Hi,

I have a crash with the following call stack:

>PxFoundationDEBUG_x64.dll!physx::shdfnd::NamedAllocator::allocate(unsigned __int64 size, const char * filename, int line) Line 99	C++
 PhysX3DEBUG_x64.dll!physx::shdfnd::Array<physx::Bp::AABBOverlap,physx::shdfnd::NamedAllocator>::allocate(unsigned int size) Line 553	C++
 PhysX3DEBUG_x64.dll!physx::shdfnd::Array<physx::Bp::AABBOverlap,physx::shdfnd::NamedAllocator>::growAndPushBack(const physx::Bp::AABBOverlap & a) Line 679	C++
 PhysX3DEBUG_x64.dll!physx::shdfnd::Array<physx::Bp::AABBOverlap,physx::shdfnd::NamedAllocator>::pushBack(const physx::Bp::AABBOverlap & a) Line 304	C++
 PhysX3DEBUG_x64.dll!physx::Bp::deleteOverlap(physx::shdfnd::Array<physx::Bp::AABBOverlap,physx::shdfnd::NamedAllocator> * overlaps, const physx::shdfnd::Array<physx::Bp::VolumeData,physx::shdfnd::NamedAllocator> & volumeData, unsigned int id0, unsigned int id1) Line 2252	C++
PhysX3DEBUG_x64.dll!physx::Bp::PersistentPairs::updatePairs(unsigned int timestamp, const physx::PxBounds3 * bounds, const float * contactDistances, const physx::Bp::FilterGroup::Enum * groups, const bool * lut, physx::shdfnd::Array<physx::Bp::VolumeData,physx::shdfnd::NamedAllocator> & volumeData, physx::shdfnd::Array<physx::Bp::AABBOverlap,physx::shdfnd::NamedAllocator> * createdOverlaps, physx::shdfnd::Array<physx::Bp::AABBOverlap,physx::shdfnd::NamedAllocator> * destroyedOverlaps) Line 2372	C++
PhysX3DEBUG_x64.dll!physx::Bp::SimpleAABBManager::updatePairs(physx::Bp::PersistentPairs & p, physx::Bp::BpCacheData * data) Line 2436	C++
 PhysX3DEBUG_x64.dll!physx::Bp::PersistentActorAggregatePair::update(physx::Bp::SimpleAABBManager & manager, physx::Bp::BpCacheData * data) Line 1028	C++
PhysX3DEBUG_x64.dll!physx::Bp::ProcessAggPairsParallelTask::runInternal() Line 2663	C++
PhysX3DEBUG_x64.dll!physx::Cm::Task::run() Line 67	C++

It looks like "volumeType" is 3 in deleteOverlap(), but the "overlaps" array it accesses has only two elements. I have a solid 100% repro for this crash in my (huge) test level, but unfortunately I don't have a simple way to repro that I could share. The crash goes away if I disable aggregates on my ragdolls.

The version I'm using is 3.4.1.25354359, but in the earlier "PhysX 3.4.1, APEX 1.4.1 Release @2284554" version I did not get this crash.

If this is not enough information to get the issue fixed, could you point me to some code I could take a look at to further investigate this?

kstorey-nvidia, 2019-01-17:
Thanks for reporting. It looks like this relates to triggers (type 3) and the problem is a direct result of this change having been back-ported from PhysX 4 and not been adjusted to reflect that PhysX 3.4 has more volume types (it has shapes, particles, cloth and triggers, whereas PhysX 4 only has shapes and triggers).

I am working on a repro and already have a speculative fix. We'll include the fix in the next release of PhysX 3.4 and I'll provide details of how to fix it once I have confirmed it.

A quick work around without modifying the PhysX source code would be to not add triggers to the aggregates.

ikkah, 2019-01-17:
Again thanks for responding so quickly! I applied those changes and the crash is now gone.

closed 2019-01-17 by ikkah