NVIDIAGameWorks/PhysX #43: disable/enable self-collision for physx::PxArticulationReducedCoordinate?

erwincoumans, 2019-01-19:
What is the recommended way to enable/disable self collisions for PxArticulationReducedCoordinate?
Is there any documentation for PxArticulationReducedCoordinate?

Posted the question here, since Adam asked for it here, instead of forum (still, I think forum has some value too): https://devtalk.nvidia.com/default/topic/1046135/physx-and-physics-modeling/please-consider-posting-questions-to-physx-github-issues-directly-/

kstorey-nvidia, 2019-01-21:
You can implement custom filtering by providing an implementation of PxSimulationFilterShader to PxSceneDesc::filterShader.

By default, you are probably calling PxDefaultSimulationFilterShader. You could probably find a way to use this filter shader to achieve what you would like by working through the code, but the simplest solution might be to implement your own filter shader function and apply your own custom rules to filter shapes.

See here https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/RigidBodyCollision.html#collision-filtering for further details.

An alternative is that you could use a broad phase aggregate to filter self-collisions. Filter shaders are a precision tool, while using aggregates to filter self-collisions is more of a blunt instrument. However, they are a simple way to achieve what you asked. You would just need to add the articulation to an aggregate with self-collision disabled.

erwincoumans, 2019-01-22:
Thanks for the reply Kier!

>> and apply your own custom rules to filter shapes.

How does one detect inside a filterShader that the objects belong to the same articulation?
~~~~~~~~~~~~~~
PxFilterFlags scissorFilter(PxFilterObjectAttributes attributes0, PxFilterData filterData0,
							PxFilterObjectAttributes attributes1, PxFilterData filterData1,
							PxPairFlags& pairFlags, const void* constantBlock, PxU32 constantBlockSize)
~~~~~~~~~~~~~~
>> broad phase aggregate

This works indeed, I had to add the aggregate to the scene, instead of the articulation.

erwincoumans, 2019-01-22:
Great, thanks a lot for your explanation. That'll do ;)

closed 2019-01-22 by erwincoumans