NVIDIAGameWorks/PhysX #44: Access to jacobians from a PxArticulationReducedCoordinate

christoph-jaehne, 2019-01-22:
Hi everyone,

I'm currently trying to figure out how to get jacobian matrices from a PxArticulationReducedCoordinate. What I try to get is the [6xNdof] matrix J_i for link i, mapping joint velocities to Cartesian velocities of link i. I followed the Documentation and so far I'm doing something like:
```
    articulation->commonInit();
    PxArticulationCache* cache = articulation->createCache();
    articulation->copyInternalStateToCache(*cache, PxArticulationCache::eALL);
    PxU32 link_idx = mylink->getLinkIndex();
    articulation->computeKinematicJacobian(link_idx, *cache);
```
With this background I have couple of questions:

- Do I access my matrix with `cache->jacobian[link_idx].j[row][col]`  or `cache->jacobian->.j[row][col]` ? I'm especially confused her as `cache->jacobian` is a pointer to an `PxKinematicJacobian` object which contains the member `xReal j[6][6]; //[column][row]`. How can both dimensions of the jacobian be hardcoded to 6? One must be the number of Dof right?
- What does the public member  `cache->jacobian->nbColumns` do? It is always zero and I can't find any assignments to it in the physx Code..
- How are the dof of floating bases handled in the jacobians?

I would really appreciate some help here as both documentation from [here](http://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/Articulations.html#inverse-dynamics) and the comments in the header `PxArticulationReducedCoordinate.h` do not really cover how to access/interpret the jacobians..

christoph-jaehne, 2019-02-06:
Any help with this?

amoravanszky, 2019-02-28:
I added a function called PxArticulationReducedCoordinate::computeDenseJacobian, it will be coming in the next update.  Please email me at amoravanszky at nvidia if you'd like to beta-test it.

closed 2019-02-28 by amoravanszky