NVIDIAGameWorks/PhysX #47: Run-Time Check Failure #2 - Stack around the variable 'bounds' was corrupted.

blab-liuliang, 2019-01-24:
In File "SqIncrementalAABBPrunerCore.cpp"
In line 396, 397

```cpp
PxBounds3 bounds;
V4StoreU(node->mBVMin, &bounds.minimum.x);
V4StoreU(node->mBVMax, &bounds.maximum.x);
```
bounds.minimum only have three float values, but why you can copy four float values to it?

AlesBorovicka, 2019-01-24:
Hi,
 no, you are right, this is a bug and needs to be fixed. 
The code should look like this:
					PxBounds3 bounds;
					V4StoreU(node->mBVMin, &bounds.minimum.x);
					PX_ALIGN(16, PxVec4) max4;
					V4StoreA(node->mBVMax, &max4.x);
					bounds.maximum = PxVec3(max4.x, max4.y, max4.z);

I am going to fix it asap, sorry for the trouble and thanks for reporting the issue!

Regards,
Ales

PierreTerdiman, 2019-01-24:
An alternative would be to introduce/use a PxBounds3Padded variation of PxBounds3 - see for example TrianglePadded, PxMat33Padded, etc, we have a bunch of these exactly for these reasons.

In this case it looks like this is jsut debug rendering code though, so anything will do

amoravanszky, 2019-02-28:
The fix for this is coming in the next update.

closed 2019-02-28 by amoravanszky