NVIDIAGameWorks/PhysX #50: Support for cylinder shapes

christoph-jaehne, 2019-01-29:
Hello everybody, as a roboticist and being used to the urdf format, I really miss the basic shape of a cylinder. Especially for wheels an approximation with e.g. capsules does not make sense as it overly expands the volume and a mesh seems like overkill for this application.

Any intentions of adding cylinders to the set of supported primitive shapes?

kstorey-nvidia, 2019-01-29:
You can use a tessellated convex mesh and get very close results to a cylinder. PhysX also supports adding a radius to any arbitrary shape (called "rest offset"), which allows you to introduce rounded edges to any convex shape, e.g. you can use this to make a lozenge-like shape from a box. Basically, you only need to use a capsule if you don't care about the accuracy of collisions at the ends.

If you want to make a cylinder shape, you should just need to sample the circles at the top/bottom of the sphere (e.g. 16 or 32 verts per-face) and pass those vertices to our convex cooking algorithm and the result should be a decent approximation to a cylinder. If you are unhappy with the rolling behaviour, you can improve that by shrinking your cylinder by some small amount and adding that value to the contact and rest offsets. This will result in slightly corners on the cylinder, but improved rolling behaviour.

christoph-jaehne, 2019-01-29:
I assume that using a mesh here instead of a geometric primitive would be computationally less efficient and also less accurate for rolling contacts as you already mentioned.

So there is no intention to add that feature in the future?

christoph-jaehne, 2019-01-29:
The videos really look promising. Meshes it is then ;) Thanks for the infos.

closed 2019-03-01 by amoravanszky

Chico63, 2020-01-07:
Hi,

I'm in charge of the development of a robotic simulator since 2010 that need to be accurate in order to be useful.
Integration of cylinder shape would be a great improvement for us in order to have better accuracy and lower cpu time consumption.
For example : when it come into simulating an IMU sensor, tessellated cylinder provide very chaotic data for the elevation as wheels always bounce to the ground whatever we tune it.
Also, it implicitly provoke slip : vehicle speed (ground speed) is lower than the target speed provided.
In order to remove these issues we sometime have to replace the wheels by spheres but it only move the problem to another place because vehicles may be larger than expected and won't respond accordingly to LIDAR sensors for example.
Don't forget that PhysX engine may be used for other things than video games.

kstorey-nvidia, 2020-01-07:
PhysX supports sphere-swept shapes, e.g. sphere-swept box (lozenge) and sphere-swept convex. It is fairly straightforward to use these - increase contact offset and rest offset on the shape, with rest offset corresponding to the amount by which you want to inflate the shape.

Chico63, 2020-01-16:
Modifying contact offset and rest offset has already been tested several times and does not improve our behavior. We need a perfect cylinder shape for our purpose, nothing less.

coffeenotfound, 2025-01-23:
Maybe as a compromise it would be great if PhysX added a utility function for generating such a cylinder mesh

vreutskyy, 2025-01-23:
The latest [PhysX SDK 5.5.0](https://github.com/NVIDIA-Omniverse/PhysX) has recently introduced [PxConvexCoreGeometry](https://nvidia-omniverse.github.io/PhysX/physx/5.5.0/_api_build/classPxConvexCoreGeometry.html#pxconvexcoregeometry). It allows simulation of cylinders, among the other quadric shapes.

coffeenotfound, 2025-01-25:
Does it really allow cylinders though? As far as I understood you could only make capsules via a line as core and a margin as radius