NVIDIAGameWorks/PhysX #51: Adding PxRigidDynamic object with PxRigidBodyFlag::eKINEMATIC motion type and PxShape with geometry type PxGeometryType::eTRIANGLEMESH crashes simulation.

lbrysh, 2019-01-31:
Hi,

When adding PxRigidDynamic object with PxRigidBodyFlag::eKINEMATIC motion type and PxShape with geometry type PxGeometryType::eTRIANGLEMESH crashes simulation.

This configuration was working fine in PhysX 3.4.

The Call Stack:
 	0000000000000000()	Unknown
>	PhysX_64.dll!physx::PxcDiscreteNarrowPhasePCM(physx::PxcNpThreadContext & context, const physx::PxcNpWorkUnit & input, physx::Gu::Cache & cache, physx::PxsContactManagerOutput & output) Line 444	C++
 	PhysX_64.dll!PxsCMDiscreteUpdateTask::processCms<&physx::PxcDiscreteNarrowPhasePCM>(physx::PxcNpThreadContext * threadContext) Line 441	C++
 	PhysX_64.dll!PxsCMDiscreteUpdateTask::runInternal() Line 514	C++
 	PhysX_64.dll!physx::Cm::Task::run() Line 67	C++
 	CarbonEditor.exe!physx::Ext::CpuWorkerThread::execute() Line 97	C++
 	PhysXFoundation_64.dll!physx::shdfnd::`anonymous namespace'::PxThreadStart(void * arg) Line 101	C++
 	kernel32.dll!BaseThreadInitThunk()	Unknown
 	ntdll.dll!RtlUserThreadStart()	Unknown

lbrysh, 2019-01-31:
This issue happened only if PXScene initialized with PxSceneDesc's members: 
kineKineFilteringMode set to PxPairFilteringMode::eKEEP 
AND / OR
staticKineFilteringMode set to PxPairFilteringMode::eKEEP

PierreTerdiman, 2019-02-01:
Thanks, I am able to reproduce a crash that looks similar.

I think it should have crashed in 3.4 as well though, if you used the PxSceneFlag::eENABLE_KINEMATIC_STATIC_PAIRS or PxSceneFlag::eENABLE_KINEMATIC_PAIRS flags in the same situation (which would be equivalent to PxPairFilteringMode::eKEEP in 3.4).

We'll work on a fix anyway.

PierreTerdiman, 2019-02-01:
I think the fix is just to remove the assert and add the if(materialMethod) in the code below:

```
	const PxcGetMaterialMethod materialMethod = g_GetMaterialMethodTable[type0][type1];

//	PX_ASSERT(materialMethod);

	if(materialMethod)
		materialMethod(shape0, shape1, context,  materialInfo);

Would you be able to try this on your end?
```

lbrysh, 2019-02-01:
Thank you for the potential fix, I'll try it in my project.
What I did to temporally band aid this issue:

//	PX_ASSERT(materialMethod);

if ( !materialMethod )
    return;

frk2, 2019-02-06:
@PierreTerdiman Thanks the above works - Not to hijack this issue (my problem seems related) but I still dont get contact reports between collisions of kinematic bodies in PxSimulationEventCallback (is that expected?). The only thing that works is to return e_CALLBACK from a custom shader and get collision pairs in PxSimulationFilterCallback::OnPairs.

frk2, 2019-02-06:
I get the warning: ```warning : Filtering: Resolving contacts between two kinematic objects is invalid. Contacts will not get resolved.```

frk2, 2019-02-06:
Ignore the above - did seem like hijacking so I opened another issue: #58

amoravanszky, 2019-03-01:
This is now fixed in our internal mainline that will be pushed during the next update.  Closing this issue.

closed 2019-03-01 by amoravanszky