NVIDIAGameWorks/PhysX #53: Link errors on Linux, same collection of libs work on Windows except PhysXTask which isn't produced.

datgame, 2019-02-01:
On Windows, I use `PhysX_64 PhysXCommon_64 PhysXCooking_64 PhysXExtensions_static_64 PhysXFoundation_64 PhysXPvdSDK_static_64 PhysXTask_static_64`
But Linux lacks the last one. All other are built, no error shown in the end.
The build process prints
`[ 34%] Built target PhysXTask`
but no such lib is available for linking. Maybe that's the reason for these:
`physxcooking/src/Cooking.cpp:97: undefined reference to `physx::shdfnd::FPUGuard::FPUGuard()'`

`pvd/src/PxPvdDataStream.cpp:784: undefined reference to physx::shdfnd::Time::getCurrentCounterValue()'`
I know ordering during linking can be a problem on Linux too. I tried switching them around a bit with no luck.

AlesBorovicka, 2019-02-01:
Hi,
 on linux you dont need to link against PhysXTask, as that project is build as an object file and linked into PhysX static library.
The undefined references above are not from PhysXTask, but from PhysXFoundation, please make sure that you link in correct order. This can indeed cause some trouble and is a source of linking issues sometimes. 
This is the linking order we have in our snippets and other internal projects that works for us:
PhysX PhysXExtensions PhysXCharacterKinematic PhysXCooking PhysXFoundation PhysXVehicle PhysXPvdSDK
Regards,
Ales

datgame, 2019-03-07:
Here's a universal "solution" I had to apply in the end anyway. Works on all projects.
(in zsh on linux)
in PhysX/physx/bin/linux.clang/debug/repack
for x in ../libPhysX*; ar x $x 
ar rcs ../libphysxrepack.a *.o
and link to that single file instead