NVIDIAGameWorks/PhysX #58: No collision contact points (only OnPairs in filtercallback) between two kinematic actors

frk2, 2019-02-06:
I dont get contact reports between collisions of kinematic bodies in PxSimulationEventCallback (is that expected?). The only thing that works is to return e_CALLBACK from a custom shader and get collision pairs in PxSimulationFilterCallback::OnPairs - but thats very early on and inaccurate. I also get a warning ```warning : Filtering: Resolving contacts between two kinematic objects is invalid. Contacts will not get resolved.``` when doing so.

The documentation doesn't state anything about this. Is this a known limitation?

kstorey-nvidia, 2019-02-07:
It isn't possible to respond to a collision between kinematic-static or kinematic-kinematic bodies because in those cases, both objects have infinite mass. Therefore, the applied force would always be infinite (and even then, the velocities of the bodies would not change).

To remove this warning, please ensure that these collisions do not raise PxPairFlag::eSOLVE_CONTACT in the filter shader.

If you require contact reports for these pairs, please ensure that you have requested them in the filter shader, e.g. by raising:

PxPairFlag::eNOTIFY_TOUCH_FOUND 
		| PxPairFlag::eNOTIFY_TOUCH_PERSISTS 
		| PxPairFlag::eNOTIFY_TOUCH_LOST 

PhysX does not report contacts unless they are requested by the filter shader.

frk2, 2019-02-07:
Thanks! I should have clarified: That works for convex mesh geometries - not for triangle mesh geometries though. Is that expected?

kstorey-nvidia, 2019-02-07:
Oh, right. Yes, PhysX does not support collisions between non-convex triMeshes. It only supports collisions between convex primitives (sphere, capsule, box and convexMesh) and between convex primitives and non-convex shapes (e.g. sphere/capsule/box/convexMesh vs non-convex mesh/heightfield/plane).

frk2, 2019-02-07:
No prob. The documentation mentions otherwise in the rigid body collisions section:

```
TriangleMesh, HeightField and Plane geometries are not supported for simulation shapes that are attached to dynamic actors, unless the dynamic actors are configured to be kinematic.
```

amoravanszky, 2019-03-01:
You can attach those shapes to kinematic actors, but the bottom line is still that until someone implements a trimesh-trimesh narrow phase function we cannot offer callbacks with results because there aren't any.  Sorry about that.

closed 2019-03-01 by amoravanszky

vreutskyy, 2025-01-06:
> Hi,I want to ask if there is a way to handle the collision of trimesh-trimesh?

This repository is for the old version of PhysX 4.1. This version doesn't support trimesh-trimesh collision.

But the newer version from [here](https://github.com/NVIDIA-Omniverse/PhysX) does support it. Check [Triangle Meshes](https://nvidia-omniverse.github.io/PhysX/physx/5.5.0/docs/Geometry.html#triangle-meshes) SDF Creation section.

hazidh, 2025-01-06:
I create two PxRigidActors which their shape is PxTriangle,how can I get the information in the PxSimulationEventCallback?I set PxSceneDesc::kineKineFilteringMode but doesn't work.
[image]
Should I use the Triangle Meshes SDF?

hazidh, 2025-02-12:
Hi I would like to ask if Gjk Queries do not support the Triangle Meshes either. The only documents I see are Spheresupport, Capsulesupport, BoxSupport, ConvexMeshsupport.

[image]

vreutskyy, 2025-02-12:
GjkQueries are for convex shapes. TriangleMesh isn't a convex shape. Though, each particular triangle of a TriangleMesh is a convex shape, so you can query potentially affected triangles first, and then do a GjkQuery with each of them. See PxCustomGeometryExt::BaseConvexCallbacks::sweep() for an example.

hazidh, 2025-02-13:
> GjkQueries are for convex shapes. TriangleMesh isn't a convex shape. Though, each particular triangle of a TriangleMesh is a convex shape, so you can query potentially affected triangles first, and [...]

Ok, thank you