NVIDIAGameWorks/PhysX #62: how to rebuild shared/static lib as MD_DynamicRelease instead of MT

hanseul1795, 2019-02-08:
Hello, 
I'm getting runtime error for MT_StaticRelease/ MD_DynamicRelease.
Which Cmake parameters should I change to build PhysX lib to MD_DynamicRelease?

Thank you!

AlesBorovicka, 2019-02-08:
Hi,
 you need to change it in the preset, so for example in:
buildtools\presets\public\vc15win64.xml 
Change:
    <cmakeSwitch name="NV_USE_STATIC_WINCRT" value="True" comment="Use the statically linked windows CRT" />

Then rerun the generate_project.bat

Regards,
Ales

hanseul1795, 2019-02-08:
> Hi,
> you need to change it in the preset, so for example in:
> buildtools\presets\public\vc15win64.xml
> Change:
> Then rerun the generate_project.bat
> 
> Regards,
> Ales

you are the best! Thanx alot!

closed 2019-02-08 by hanseul1795