NVIDIAGameWorks/PhysX #64: Experience with compatibility with MS HoloLens or UWP application

liedtkeInTUM, 2019-02-12:
I'm trying to deploy the Physx Dlls with an app for the MS HoloLens. But it fails loading the app because of missing dlls. Until now I have no clear idea if the dlls are not deployed to the hololens or if there are other dependencies introduces by the PhysX Library that can't match.

If I use it in a "normal" uwp app, everything works fine. But as soon as I want to start it on the HoloLens (or emulator) the app launch fails. As "The program '[4840] HolographicApp3.exe' has exited with code -1073741515 (0xc0000135) 'A dependent dll was not found'" is very specific I just try to resolve this with a dependency walker but no success so far...

So if someone of you have already successfully deployed an app with PhysX as Hololens project, please let me know.

AlesBorovicka, 2019-02-12:
Hi,
 the required dlls should be PhysXFoundation, PhysX, PhysXCooking.
Also you can try to change the compilation of PhysX to static libraries and link directly those static libraries. 
You can change this in a preset you used -> they are stored in buildtools/presets/public.
So for example in:
buildtools\presets\public\vc15win64.xml
Change:
cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="False" comment="Generate static libraries"

Then you can link directly against the static libraries. PhysX itself does not require any other dlls, PhysXGpu on the other hand, if used, will require PhysXDevice and cuda dlls.

Regards,
Ales

liedtkeInTUM, 2019-02-13:
Hi Ales,

thx for the fast reply.  I already used static libraries for a normal win32 build, that worked. But for Universial Windows Projects the runtime library needs to be /MD(d) for dynamic linking. This can be also with the switch NV_USE_STATIC_WINCRT set to false. That was also needed for building a DLL that can be  linked from a UWP app.

So far, I tried only combinations with dlls and different wincrt settings. Not with libs, although I use this combination for all other projects included in my solution. So I just tried this. 
So combination PX_GENERATE_STATIC_LIBRARIES=True and NV_USE_STATIC_WINCRT=false works also as win32 app. But now linked from an universial app, I get "cannot open file 'delayimp.lib'".
But I hope this should be somehow fixable. I have to use following defines for static libs, right?
```
#define PX_PHYSX_STATIC_LIB
#define PX_FOUNDATION_DLL 0
```

AlesBorovicka, 2019-02-13:
In the latest code you need to either have:
 #define PX_PHYSX_STATIC_LIB
or include generated PxConfig.h that does have this define included.

As for the delayimp lib, that is something that should not be required for static libs indeed, I can take a look at that later. It would require to remove all the delay load stuff from linker at least. 
That should be in the windows cmake files for PhysXCommon, PhysXCooking and PhysX.

liedtkeInTUM, 2019-02-13:
I just started to change code in the SDK to fix unresovled externals with removed delayimp.lib linkage.
There seems to be no one to one replacement for universial projects.
So I think there is no way around source code modifications...
Found that https://chromium.googlesource.com/chromium/src/+/c5686578e8f519eeec50f574b284fbc12e6e15d8 with hint on dloadhelper, but no luck.

thx for your advise. If you have some good idea to avoid these load dll calls, please let me know.

AlesBorovicka, 2019-02-15:
Hi,
 I have removed the delay loading code (and the library requirement) for static library configurations, the change should be available most likely next week.
Regards,
Ales

liedtkeInTUM, 2019-02-18:
Hey,
that sounds good. I have just tested some hacks to exclude shared stuff from libs and discovered some additional pitfalls.
1. I need to disable CUDA (as it would also not work on hololens), but the compiler flag can't be set with cmake switches. I have a small commit for that issue, can create merge request from e97f435b4cea376d4b74ebec6b24b15a7c42b710
2. It seems that "PX_GENERATE_GPU_STATIC_LIBRARIES" with true is also needed as cmake switch
3. After exclude GPU and DLL stuff there are only two lines in the thread impl, that still conflicts with UWP :/ As UWP wants you to do everything as task, this might be a complete different implementation... for now I can hack this and have valid libs that can be linked against (79edb2e2caef9fdb8382dee389bd653087822bc8). There is also a small wchar adaption needed in PhysXIndicator...
4. The config file isn't copied with install target, right? Looks like it is only generated in source directory. Is that intended?

Best,
Sven

AlesBorovicka, 2019-02-19:
I would recommend using a new platform cmake files for that. I did ported the sdk once to uwp completely, but that was some time ago. 
As for the sdk itself I can take a look if the uwp projects still work. There are indeed some changes needed for threading. I can take a look next week and send you the cmake files. But a full support for uwp won't happen most likely...

liedtkeInTUM, 2019-02-21:
Thanks a lot for the support so far.
Right now I use following platform file to build PhysX as Universial Build (https://gist.github.com/liedtkeInTUM/de81be1c5cd4d6dfcc890fa6926391a7).
I'm going to wait on the next update and see if I can remove my modifications. Otherwise I will report new Issue for the linking error when using PX_GENERATE_GPU_STATIC_LIBRARIES=True as it also happens on normal win32 build or send a merge request to disable cuda

AlesBorovicka, 2019-02-26:
ok the next update should have the uwp support included. There should be a new preset that will create uwp projects. We will try to push the next update around next week most likely. We wont support snippets though with uwp, just projects for the sdk will be available.

closed 2019-03-01 by amoravanszky