NVIDIAGameWorks/PhysX #65: NpWindowsDelayLoadHook.cpp and CmWindowsDelayLoadHook.cpp duplicate?

erwincoumans, 2019-02-13:
Those files have the same symbols, should one of them go?
~~~~~~~~~~~~~~~~
PhysX/physx/source/physx/src/windows/NpWindowsDelayLoadHook.cpp
PhysX/physx/source/common/src/windows/CmWindowsDelayLoadHook.cpp

FARPROC WINAPI delayHook(unsigned dliNotify, PDelayLoadInfo pdli)
PfnDliHook __pfnDliNotifyHook2 = delayHook;
~~~~~~~~~~~~~~~~

AlesBorovicka, 2019-02-14:
No, I dont think so. One is in PhysX library and one in PhysXCommon library, so they should end up in a different dll's. 
Where PhysX does delay load PhysXFoundation and PhysXCommon, while PhysXCommon does delay load PhysXFoundation.
Please, where do you see issues with that?
Regards,
Ales

erwincoumans, 2019-02-14:
The issue is when you link all PhysX files statically, you end up with duplicate symbols. It would be nice to use unique names for them. Not urgent, feel free to close this.

AlesBorovicka, 2019-02-14:
I see, with static linking that code should not be there at all I guess. I will take a look.

closed 2019-02-14 by AlesBorovicka