NVIDIAGameWorks/PhysX #66: Determinism across different architectures?

tysoni, 2019-02-15:
I have an issue where PhysX simulations are returning different results when run on machines with different numbers of CPU cores. My code base is pretty complex so I'm trying to rule out possibilities and determine if the issue is related to PhysX itself, or part of my own integration of it.

If I run an identical simulation (with multithreading enabled) on machines with different CPU core counts, is it expected that I will get identical results? Or does PhysX multithreading only guarantee determinism during successive runs on the same machine?

kstorey-nvidia, 2019-02-15:
It is expected to be deterministic regardless of the number of cores and we've verified internally that we get the same results running the same executable using different numbers of threads. 

However, it is expected there would be differences betweendifferent platforms, e.g. a win32/win64/Linux build may produce different results due to differences in compilers. Similarly, a win64 debug and win64 release build may produce different results due to differences in code produced by the compiler. Unfortunately, floating point maths rounding depends on some compiler decisions.

Could you give details about what the 2 machines are? Is one AMD and one Intel, by any chance? It would be interesting if there was a subtle difference between different vendor's CPUs.

cmp-, 2019-02-15:
Is PxSceneFlag::eENABLE_ENHANCED_DETERMINISM required for it to be deterministic regardless of the number of cores? Was that enabled?

kstorey-nvidia, 2019-02-15:
No, that flag should only be required to ensure that adding additional actors to a scene doesn't change the simulation outcome of the existing actors in the scene when you rerun the simulation.

Determinism also depends on having everything happen in the same order - actor creation, same time-steps, same api calls made on the physics engine on the same frames. You also have to fully destroy and recreate the scene to ensure determinism due to how the physx scene pool allocates some objects.

tysoni, 2019-02-15:
Thank you for the quick response. I haven't done a deep dive into my code to figure out exactly why things are different across machines, was just doing some preliminary research to figure out if this is a known issue with PhysX before starting the full debugging process. Everything in my setup is created in the same order, with the same properties...however, it's perfectly plausible that the error is in my code.

I believe all the machines are Intel CPUs, but I will double check that. I will also report back once I determine what the issue is. Cheers!

closed 2019-03-01 by amoravanszky

IronWarrior, 2020-10-20:
@kstorey-nvidia this has been closed for awhile, but I am interested if PhysX determinism has ever been tested between vendors (as you mentioned above). I am building a game in Unity that uses PhysX and have [achieved full determinism for our replay system](https://twitter.com/roystanhonks/status/1318677036177784835) for same build/different machines, but only so long as the vendors are the same. Using Unity of course means there is a much larger tech stack than PhysX, but I figure it would be nice to cross that off the list if it has been tested.