NVIDIAGameWorks/PhysX #67: Source for PhysX DCC plug-ins

awefers, 2019-02-15:
I was wondering if it would be possible to release the source for the Maya and 3DS MAX DCC plug-ins? Either with the SDK source or in a separate repository in case the licenses are not compatible?

amoravanszky, 2019-03-01:
Thanks for the tip.  I will look into whether we can do this.

closed 2019-03-01 by amoravanszky

rushgreen74, 2021-02-08:
Did this get actioned? Is the DCC source code available to build plugins for Maya 2020?

awefers, 2022-07-28:
Note: The frustrating part is, that the last released version of the Maya plug-in is for Maya 2018 (or LT 2019). The problem is that Maya 2018 is now no longer available for download. It would be great to see either updated plug-in binaries or a source release. A bit of shame to let this tech all slowly disappear.