NVIDIAGameWorks/PhysX #68: Download/Build PhysX 4.0 SDK Binaries

TomerZeitune, 2019-02-16:
generate_projects.bat does not generate a visual studio sln file that can be used to build PhysX 4.0 . 
No binaries are available for download anywhere. Can anyone give a download to binaries or to a working sln file and proj file ? Why not  give binaries of the SDK on a major release like 4.0? or at least update the scripts to make an solution file to work with 4.0?

AlesBorovicka, 2019-02-16:
I am not quite sure that I understand your issue.
We do not ship binaries anymore since the source code is now here on github.
What step does not work for you when you generate your project files?

 * Solutions are generated through a script in physx root directory: generate_projects.bat
 * Script generate_projects.bat expects a preset name as a parameter, if a parameter is not provided it 
    does list the available presets.
 * Supported presets for windows platform are: vc12win32, vc12win64, vc14win32, vc14win64, vc15win32, vc15win64.
 * Generated solutions are in folder compiler/"preset name"

TomerZeitune, 2019-02-20:
OK the only sln file I found was 
PhysX-4.0\physx\compiler\vc15win64\CMakeFiles\3.13.2\CompilerIdCXX.sln
When I opened it in visual studio and runned nothing happend. Am I miss using it ?

liedtkeInTUM, 2019-02-21:
the sln file you need for build all libs you should find in PhysX-4.0\physx\compiler\vc15win64\PhysXSDK.sln. If not, your generation process seems to be broken. For me it works following that http://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/BuildingWithPhysX.html#generating-projects-with-cmake guide

closed 2019-03-01 by amoravanszky

wilflalonde, 2019-09-10:
After running generateProjects.bat, getting file PhysXSDK.sln, and building successfully
without errors, I have 8 files in bin of the form win.x86_32.vc141.mt (141 doesn't mean anything
to me). Only this one has libs in it. I'm failing to link because the linker is looking for 
PhysX3CHECKED_x86.lib which I believe is a lib from an old version of PhysX.  It's not in the 
libs with vc141.mt and all my libs are together in one directory. The project's properties don't 
reference any other location for libs. So it must be coming from one of the generated libs. 
So far, I'm stumped...

AlesBorovicka, 2019-09-11:
Yes, this has changed. The produced libs do not have the configuration name appended. The libraries have same name and are in different directories for each configuration. 
The number in the directory name are Visual studio toolset versions, note that in your case you used Visual Studio 2017 that does have the toolset version 141.

wilflalonde, 2019-09-11:
Thanks very much... I finally found my problem. My BAD.

As for the other person who was having problems too, it looked
like the same problem I had earlier, namely that MAKE was not
properly installed which I figured out by executing "make -version".
It essentially told me to re-install.

Wilf

PcChip, 2019-09-20:
I had to delete the @call CLEAN_EXIT from the top of the batch file to make it work for me. Now it tells me that CMake is missing, which is an error I can certainly deal with