NVIDIAGameWorks/PhysX #69: Modeling an aircraft on a flexible cable

xgerrmann, 2019-02-17:
##### Introduction

I wish to model a certain system with the help of a physics simulation engine. In my search, I encountered several engines amond PhysX. Currently, it is not clear whether PhysX will suit my requirements, therefore I have several questions. First I will describe my system and simulation requirements after which I pose my questions.

##### System description

The system consists of two components:

1. A flexible cable
2. An aircraft

The flexible cable is the connection between the ground and the aircraft (it is an airborne wind energy system). The cable's length should be able to change (reeling). The aircraft will be modeled as a rigid body. Cables are often modeled as a series of lumped masses connected by spring-damper elements, or by inelastic joints. I wish to model a cable of a length of 300 meters by 5 (ideally more ~20) lumped masses. The cable elements and the aircraft are subjected to aerodynamic forces. Reeling is performed by the addition and removal of elements at the bottom of the cable.

Right now I am considering modeling the cable as a lumped-mass system through particle dynamics, thus ignoring the orientation of each lumped mass. This should reduce the computational complexity. However, I cannot find a system which combines particles with rigid bodies.

##### Summary of requirements

- The creation + destruction of cable elements/ particles.
- A connection between the aircraft + cable.
- External application of additional forces to each rigid body +/ particle during simulation.
- Access to information such as position and pose during simulation.

##### Questions

I have the following questions

- Can I use a combination of particles and a rigid body?
- If not, how could I best model this system?
- Could Nvidia PhysX be capable of simulating the system as fast as real-time and with a sufficient update rate >20Hz?
- Are there any other recommendations/ remarks?

I wish somebody can help me with finding an answer to these questions. Finding a simulation engine that meets the requirements will possibly enable further work in this field of research with the same engine.

lukehb, 2019-02-24:
I'm not a Physx dev, but I have used Physx in games. Regarding your questions:

1. **Can I use a combination of particles and a rigid body?** Perhaps you could. But for a flexible cable this is typically modelled in Physx using "articulations".
2. **If not, how could I best model this system?** Use articulations to model the cable and attach the cable to the aircraft using a normal Physx joint. In Physx 3.4 they provided a code snippet that demonstrates a flexible cable with a heavy attached weight on one end, this might be a good starting point for you. [See here.](https://github.com/NVIDIAGameWorks/PhysX-3.4/blob/master/PhysX_3.4/Snippets/SnippetArticulation/SnippetArticulation.cpp)
3. **Could Nvidia PhysX be capable of simulating the system as fast as real-time and with a sufficient update rate >20Hz?** With a sufficient update rate Physx simulation should be capable of this. Physx supports sub-stepping if even further physics step granularity is required.
4. **Are there any other recommendations/ remarks?** I am working on a project with articulations currently, but the reeling in and out feature you mention has been a little difficult for me to achieve ([see here](https://github.com/NVIDIAGameWorks/PhysX/issues/74)). In terms of applying force to the cable bodies and querying their position and pose this is all possible in Physx.

xgerrmann, 2019-02-25:
@lukehb Thank you for your answer! I see that the [issue](https://github.com/NVIDIAGameWorks/PhysX/issues/74#issue-413791682) has been resolved as well so that even more shows the feasibility. 
Would you mind sharing a screen record of what you made?

lukehb, 2019-02-26:
I was using articulations to make a fishing line that can be cast and reeled in. I have turned on the debug drawing so you can see the capsule bodies that make up the fishing line articulation.

Image below is a link to the youtube video of it.

[image]

closed 2019-04-26 by xgerrmann