NVIDIAGameWorks/PhysX #70: Reduced coordinate articulation joint drives affect joints further down the chain on collision

nov-ta, 2019-02-22:
I'm trying to set up a reduced coordinate articulation using joint drives. I'm currently trying to figure out how to tune this correctly, as I'm seeing some behavior that is unexpected to me.

If the articulation's movement is blocked by a static object, it seems that joints that should have no drive force applied still move in the direction of the driving joint. This is especially visible if the joint stiffness and maxforce is maxed out, which could be an invalid configuration, but it's still occuring to a lesser degree even if they're not.

Any pointers on how to properly set this up is much appreciated! Below are two animations of the behavior I'm seeing (they're not looping perfectly, so don't pay attention to behaviour when it's not colliding with the flying box), and the code.

Stiffness/max force maxed out:
[image]

Stiffness/max force as in code listing:
[image]

Here is the code for the version with the smallest outer joint movement. It's based on the original SnippetArticulation.cpp, and should be a drop-in replacement for it.

Reducing the density or setting the maxForce/stiffness to PX_MAX_F32 will cause the behavior to become like in the first gif.

*SnippetArticulation.cpp*

```
#include <ctype.h>
#include <vector>

#include "PxPhysicsAPI.h"

#include "../snippetutils/SnippetUtils.h"
#include "../snippetcommon/SnippetPrint.h"
#include "../snippetcommon/SnippetPVD.h"

using namespace physx;

// Kept around to keep SnippetArticulationRender.cpp unaltered.
bool gCreateLiftScene = true;

const float RadToDeg = 180.0f / PxPi;
const float DegToRad = 1.0 / RadToDeg;

PxDefaultAllocator		gAllocator;
PxDefaultErrorCallback	gErrorCallback;

PxFoundation*			gFoundation		= NULL;
PxPhysics*				gPhysics		= NULL;

PxDefaultCpuDispatcher*	gDispatcher		= NULL;
PxScene*				gScene			= NULL;

PxMaterial*				gMaterial		= NULL;

PxPvd*                  gPvd			= NULL;

PxArticulationReducedCoordinate*		gArticulation = NULL;
PxArticulationJointReducedCoordinate*	gDriveJoint = NULL;

PxFilterFlags articulationFilter(PxFilterObjectAttributes attributes0, PxFilterData filterData0,
							PxFilterObjectAttributes attributes1, PxFilterData filterData1,
							PxPairFlags& pairFlags, const void* constantBlock, PxU32 constantBlockSize)
{
	PX_UNUSED(attributes0);
	PX_UNUSED(attributes1);
	PX_UNUSED(constantBlock);
	PX_UNUSED(constantBlockSize);
	if (filterData0.word2 != 0 && filterData0.word2 == filterData1.word2)
		return PxFilterFlag::eKILL;
	pairFlags |= PxPairFlag::eCONTACT_DEFAULT;
	return PxFilterFlag::eDEFAULT;
}

void createArticulation()
{
	gArticulation = gPhysics->createArticulationReducedCoordinate();

[... 215 lines cut ...]
}

int snippetMain(int, const char*const*)
{
#ifdef RENDER_SNIPPET
	extern void renderLoop();
	renderLoop();
#else
	static const PxU32 frameCount = 100;
	initPhysics(false);
	for(PxU32 i=0; i<frameCount; i++)
		stepPhysics(false);
	cleanupPhysics(false);
#endif

	return 0;
}

void keyPress(unsigned char key, const PxTransform& camera)
{
	PX_UNUSED(key);
	PX_UNUSED(camera);
}

```

kstorey-nvidia, 2019-02-28:
Thanks for reporting this issue.

The issue relates to the order in which constraints are solved. The joint drives are solved before the contacts, applying torques to drive to the target velocities. This makes both links rotate. Following this, a contact is solved which stops the first link from rotating, but the second link continues to rotate.

If there is a physical limit on the torque that the motors can apply, then this process will eventually reach a point where the system converges, otherwise the drive continues to set a target velocity that the contact solver overrules.

We have an improvement to the articulation solver that will be made available in the near future that handles this case correctly. The new solver is also significantly faster than the current code. The following are the results from just 1 iteration. The results in this case are identical with both the PGS and TGS solvers:

[image]

These improvements should be made available very soon but, in the meantime, applying motor limits should be a reasonable workaround.

We'll let you know when this feature becomes available

Thanks

Kier

nov-ta, 2019-03-01:
Thanks for the update, that does look promising!

closed 2019-03-01 by amoravanszky