NVIDIAGameWorks/PhysX #72: Constraints start to fail when rigidbody count exceeds 65536 (PxU16)

tysoni, 2019-02-22:
I have an offline simulation with a lot of rigidbodies connected with constraints (aiming for 100k+ RBs). I noticed their constraints start to fail miserably once the number of rigidbodies exceeds 65536. 

I turned on Px debugging and found the failure is happening in DyDynamics.cpp (line 329), because the bodyADataIndex (presumably the RB index for a constraint) is being cast to a 16 bit integer, so an overflow is occurring when an RB index exceeds that value.

Is there a specific reason for this, and is there a workaround?

kstorey-nvidia, 2019-02-22:
The storage type is a PxU32, so you should be able to remove the cast in that code and it should work. We used to store in a PxU16 type, but this was switched to a PxU32 specifically to remove the 64k body limit. Could you try removing the Ps::to16() on line 329 and 348 and see if this resolves your issue?

The code that appears to have not been updated relates to joints. I have run scenes with 100k boxes successfully, so AFAIK just removing those truncations should be sufficient.

tysoni, 2019-02-22:
Thanks, I figured that was the solution...but wanted to ask here first before changing anything.

I've re-run the sim with those casts removed and the problem no longer seems to occur.

closed 2019-02-22 by tysoni