NVIDIAGameWorks/PhysX #73: Python Bindings for PhysX

alanjfs, 2019-02-23:
Is there any interest or existing initiative to bring PhysX closer to Python, especially in the context of machine learning?

alanjfs, 2019-02-24:
Getting the ball rolling on this.

- https://github.com/alanjfs/PyPhysX4

```python
import time
import PhysX4 as px

if __name__ == '__main__':
    px.initPhysics()

    px.createPlane(
        px.PxPlane(0, 1, 0, 0)
    )

    px.createStack(
        px.PxTransform(px.PxVec3(0, 0, 10.0)),
        size=5,
        halfExtent=2.0
    )

    px.createDynamic(
        px.PxTransform(px.PxVec3(0, 50, 100)),
        geometry=px.PxCapsuleGeometry(radius=5, halfHeight=5),
        velocity=px.PxVec3(0, -50, -100)
    )

    print("Simulating..")

    t0 = time.time()
    steps = 100
    timestep = 1.0 / 30
    for i in range(steps):
        px.stepPhysics(timestep)

    t1 = time.time()
    duration = t1 - t0

    px.cleanupPhysics()

    print(
        "Finished in %.2f ms (%d fps)"
        % (duration * 1000, steps / duration)
    )
```

The above is working well, and is more or less ported from [SnippetHelloWorld](https://github.com/NVIDIAGameWorks/PhysX/blob/4.0/physx/snippets/snippethelloworld/SnippetHelloWorld.cpp), with parts remaining in C++ and other parts being called from Python.

The issue I ran into however was with regards to memory management. I'm not very well versed with C++, and these functions all carefully avoid the use of PhysX's [PxAllocatorCallback](http://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/API.html#memory-management); that's where my understanding starts to fade. From what I gather, PhysX manages the lifetime of objects internally and does reference counting on its own, similar to (I expect) how Python would normally do things. Spontaneously, I'd imagine a marriage between these to fit hand in glove, but I'm not sure how to make it happen.

Ultimately I'd like to implement all of SnippetHelloWorld in Python, like this:

- https://gist.github.com/alanjfs/cf511e96751068351a34be5d42edc506

But I ran into a few issues, of particulat interest is:

```python
def createDynamic(t, geometry, velocity):
    dynamic = px.PxCreateDynamic(gPhysics, t, geometry, gMaterial, 10.0)
    dynamic.setAngularDamping(0.5)
    dynamic.setLinearVelocity(velocity)
    gScene.addActor(dynamic)
    return dynamic
```

This doesn't work at the moment, because `PxCreateDynamic` is a factory function, returning a `PxRigidDynamic` instance which in turn has a protected destructor, and thus won't let Python (possibly more specifically pybind11) manage its lifetime.

Things I'm considering is either declaring a [custom constructor](https://pybind11.readthedocs.io/en/master/advanced/classes.html#custom-constructors) or a [custom smart pointer](https://pybind11.readthedocs.io/en/stable/advanced/smart_ptrs.html#custom-smart-pointers), both of which are out of my depth.

Any ideas?

totoze, 2019-02-26:
Hello , I would like to contribute to the python binding too. I have an experience with boost.python in cpp.
What should I do ?

alanjfs, 2019-07-22:
Hi @totoze, that would be great!

1. You can start having a look at [main.cpp](https://github.com/alanjfs/PyPhysX4/blob/master/cpp/src/main.cpp), the project is this single file and is quite short so far. 
1. Once you've managed to get that built, I would try porting more of the official examples to Python until each of them runs.
1. At that point, I would have a look at trying to reduce the size of the Python examples, perhaps by making some of the calls more Pythonic if necessary.
1. At that point, we've got most high-level features implemented and could potentially start applying some of MuJoCo's and PyBullet's Deep Learning examples on it, to see what's missing in terms of features and stability.
1. At this point I expect there to be a few things to optimise, both in terms of performance and workflow. It would probably be a good jumping off point for an official release, to get some more hands and ears on it.

petrikvladimir, 2020-05-06:
Hi, since @alanjfs project is not maintained anymore, I would like to point you to my recently created python wrapper located here: https://github.com/petrikvladimir/pyphysx

It is not yet very advance but might be a good starting point for adding new capabilities - I will be including new features as I will need them.
Features:
  - provides an interface to rigid actors (static, dynamic) and to shapes (box, sphere, or convex mesh)
  - provides an interface to D6Joint (no articulation yet)
  - provides an interface to Scene
  - visualizes the scene using pyglet (therefore, it is easy to modify visualization at python side)

Any comments/suggestions/issues and pull requests are warmly welcomed.