NVIDIAGameWorks/PhysX #74: Dynamically adding/removing articulation links at runtime results in crash (Physx 3.4 and 4.0)

lukehb, 2019-02-24:
1. Create an articulation
2. Fill it with 20 links 
3. Add the articulation to the scene
4. Over time remove leaf links until only the root link remains (calling link->release() on the leaf link).
5. Over time add 63 links
7. Repeat 5/6 until a crash occurs.

Additional information: 
- Crash goes away if the articulation is made initially with 64 links. But is replaced with a problem where new links are seemingly asleep and calling wakeup causing an assertion crash because links 2-64 are not active.
- Calling wakeup on the articulation when adding new links at runtime crashes straight away.
- It seems like internally the articulation does not manage the removal of links entirely. Inspecting with debugger will reveal that the data structures holding links, joints, and bodies have far more elements than they should have after links have been released.
- I have attached a drop-in reproduction that is a modification of SnippetArticulation.cpp from Physx 3.4.
- I also was able to reproduce this in 4.0 too, but 3.4 is the one I have to use because of UE4.

[Gist here](https://gist.github.com/lukehb/3921351abc2a4a60f99ea3968ccdd7c8)
[SnippetArticulation34.zip](https://github.com/NVIDIAGameWorks/PhysX/files/2897704/SnippetArticulation34.zip)

Is this my misuse of the SDK or is it a bug?

michellelu-nvidia, 2019-02-25:
Hi Luke,

Thanks for reporting this issues. I have a fix for it. Please go to removeBody(BodySim& body) method in ScArticulationSim.cpp, after removing the link, we need to remove the corresponding body and joint as well.

	mLinks.popBack();
	mBodies.popBack();
	mJoints.popBack();

Please let me know whether this fix your issues.

Many thanks

Michelle

lukehb, 2019-02-25:
Hi Michelle, 

I have tested your fix and it works. I really do appreciate your quick response and fix - this was a quite the blocking bug for us.

Thanks again,

Luke.

michellelu-nvidia, 2019-02-25:
Hi Luke,

Np. The fixes will be in our next GitHub release on both 3.4 and 4.0.

Thanks for reporting it.

Cheers
Michelle

closed 2019-02-26 by lukehb