NVIDIAGameWorks/PhysX #75: Failed linking of physX.

totoze, 2019-02-26:
Hello , I'm trying to link physX for sometime now. as the title suggest I'm unsuccessful so far. I read the guide countless times and can't figure out what's wrong but I have a feeling I might've messed up the build configuration while building the sdk. 
Here's the information:
ERROR i'm getting : 
_error LNK2019: unresolved external symbol __imp__PxCreateFoundation referenced in function "public: __thiscall PhysicsSystem::PhysicsSystem(void)" (??0PhysicsSystem@@QAE@XZ)_

code (note ESC\System.h is engine specific ignore it):
```
#pragma once
#include <pch.h>
#include "PxPhysicsAPI.h"
#include "ECS\System.h"

using namespace physx;

class Allocator : public PxAllocatorCallback
{
public:
	~Allocator() {}
	void* allocate(size_t size, const char* typeName, const char* filename, int line) {
		return _aligned_malloc(size, 16);
	}
	void deallocate(void* ptr) {
		_aligned_free(ptr);
	}
};

class ErrorReporter : public PxErrorCallback
{
public:
	void reportError(PxErrorCode::Enum code, const char* message, const char* file,
		int line)
	{
		std::cout << "Hello from physX \n" << message <<
			"\n At File:" << file << "\n At Line:" << line << std::endl;
	}
};

class PhysicsSystem : public System {
private:
	Allocator gAllocator;
    ErrorReporter gReporter;
	PxFoundation*			mFoundation = NULL;
	PxPhysics*				mPhysics = NULL;

	PxDefaultCpuDispatcher*	mDispatcher = NULL;
	PxScene*				mScene = NULL;
	PxMaterial*				mMaterial = NULL;
public:
	PhysicsSystem() 
	{
		mFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gAllocator,gReporter);
		if (!mFoundation)
			printf("PxCreateFoundation failed!");
	}
};
```

Linker->Input->Additional Dependencies :
PhysXCommon_64.lib
PhysX_64.lib
PhysXFoundation_64.lib
PhysXExtensions_static_64.lib
Linker->General->Additional Library Directories:
$(ProjectDir)Dependencies\PHYSX\lib
C/C++->General->Additional Include Directories:
$(ProjectDir)Dependencies\PHYSX\include
Build Confing:
Debug x86
Warnings:
PhysXCommon_64.lib : warning LNK4272: library machine type 'x64' conflicts with target machine type 'x86'
[... 370 lines cut ...]
        SnippetUtils_static__64.pdb
        SnippetVehicle4W_64.exe
        SnippetVehicle4W_64.ilk
        SnippetVehicle4W_64.pdb
        SnippetVehicleContactMod_64.exe
        SnippetVehicleContactMod_64.ilk
        SnippetVehicleContactMod_64.pdb
        SnippetVehicleMultiThreading_64.exe
        SnippetVehicleMultiThreading_64.ilk
        SnippetVehicleMultiThreading_64.pdb
        SnippetVehicleNoDrive_64.exe
        SnippetVehicleNoDrive_64.ilk
        SnippetVehicleNoDrive_64.pdb
        SnippetVehicleScale_64.exe
        SnippetVehicleScale_64.ilk
        SnippetVehicleScale_64.pdb
        SnippetVehicleTank_64.exe
        SnippetVehicleTank_64.ilk
        SnippetVehicleTank_64.pdb
```
And my sdk build config is vc14win64 on Debug found at physx\bin\win.x86_64.vc140.mt\debug
PhysX version 4.0.0 using it in a custom game engine.

I have no idea what's wrong but hope there's enough information to find the problem. 
Thank you for your time.

AlesBorovicka, 2019-02-26:
Looks like you are using x64 libraries to link against your x86 app:

totoze, 2019-02-26:
> 
> 
> Looks like you are using x64 libraries to link against your x86 app:
> PhysXCommon_64.lib : warning LNK4272: library machine type 'x64' conflicts with target machine type 'x86'
> 
> Build and link against x86 libs, use preset vc14win32.

Thanks a lot. that + moving the dlls to the same folder as my exe fixed the issue.

closed 2019-03-01 by amoravanszky