NVIDIAGameWorks/PhysX #76: Missing CMake error when running generate_projects.bat windows

JDAlzate, 2019-03-01:
I'm trying to generate the projects for PhysX but I keep getting the same error: "Cmake is missing, please install cmake version 3.12 and up."
The error is pretty obvious but I have already installed CMake.
I'm not sure if there is a specific directory I should install CMake on, I have never worked with CMake before. Right now, it's installing in the default program files folder.

nov-ta, 2019-03-01:
It probably has to be available on path, try running "where cmake" from a cmd window and see what pops up. That's what generate_projects.bat uses to find cmake. In my case it's part of user path:

[image]

JDAlzate, 2019-03-04:
I'm not sure what the problem was. I just uninstalled CMake and installed it again and it worked just fine.
Thank you for your help though, I appreciate it.

closed 2019-03-04 by JDAlzate