NVIDIAGameWorks/PhysX #77: Simulate a single actor, assuming all other actors are kinematic?

StormtideGames, 2019-03-02:
I realise that this isn't something PhysX would normally allow you to do (and I think I can understand the reason why it's not done) - but I'm wondering if someone can point me in the right direction for implementing this.

I'm using PhysX with Unreal Engine 4 to build a medium-scale game with a heavy focus on vehicles. These vehicles have custom methods of applying forces to achieve a specific style of movement, and I would like to use PhysX to integrate those movement forces and resolve collision with the rest of the scene.

The game is multiplayer, and the scenes that the vehicles are part of are quite large. I need to have a strict Server-Authoritative simulation with client-side prediction, but for that to be feasible I need to be able to simulate individual actors at specific times immediatelly. Clients need to replay a history buffer of movement if they receive a correction from the Server. This would currently require duplicating the scene at least once on each client, and once for _each_ client on the Server - this will quickly become unmanageable and costly.

It is straightforward to pause the simulation of a single dynamic actor, by marking it as kinematic and saving it's physics state. I want to be able to do the _inverse_ of this:

- Mark all objects in the scene as temporarily kinematic.
- Mark a single kinematic actor as dynamic.
- Simulate the scene (and therefore the only dynamic actor), and retrieve the resolved state of the dynamic actor. The simulation will accumulate any collision forces against the temp-kinematic actors.
- Clear the 'temporary kinematic' flag, restore the kinematic flag on the single dynamic actor.
- Continue scene simulation as normal.

This would essentially allow me to simulate a single actor at a time using the most up-to-date version of the scene, and resolve collisions for that actor as if all other actors are kinematic. This removes the need for any scene duplication and should provide an acceptable method for replaying prior movements on clients.

Is this achievable, or am I barking up the wrong tree entirely? I understand there are other approaches to validating client-side prediction, but none are accurate enough for my needs.

kstorey-nvidia, 2019-03-04:
I think this is a job that would be best handled using PhysX immediate mode. This is a low-level API that gives you direct access to collision detection, the solver and integration. Immediate mode is a stateless set of C functions, does no internal threading/work batching and, as such, is very low-overhead when simulating very simple scenes (e.g. a single object moving around the world). It is commonly used for embedded simulations as part of character animation, and also as a way to quickly replay simulations in networked games.

The drawback with immediate mode is that the API is a much lower-level API and requires the application code to organise data and group objects together for collision detection or the solver to process. It is just a handful of data structures and utility functions which you assemble to put together your own embedded simulator.

StormtideGames, 2019-03-05:
Thanks - I have had a look into immediate mode, and UE4 does have an implementation of this in terms of using it within a character animation system, so getting to grips with it should be fairly straightforward.

That said, immediatte mode still requires a lot of duplication of data from the main scene - and I have a high number of dynamic objects all of which can interact with each other. Because of this, I would like to avoid duplication into an immediatte scene - and instead "pause" other actors in the main scene, while still allowing them to accumulate collision resolution forces for when they are "unpaused".

If that is not possible without some considerable reworking of PhysX, then perhaps immediate mode is indeed the most appropriatte solution. Any examples for this would be much appreciatted if available!

kstorey-nvidia, 2019-03-05:
Feel free to do it whichever way works best for you.

Please keep in mind that converting from dynamic -> kinematic will do what you want, but it will lose any velocity information and put the kinematic to sleep. When converting back to dynamic from kinematic, you will need to have stored the dynamic state of the actor (velocities, whether it was awake) and reset these values on the dynamics, so you'll need to store that state somewhere. 

Converting from dynamic->kinematic and back again will involve re-filtering all interactions and potentially a re-insertion into the broad phase, so it can get quite expensive if you do it for a large number of bodies with a large number of interactions. It will also drop any cached information so it could subtly change the outcome of the simulation (this one probably isn't a huge deal but it will cause problems if you depend on deterministic simulation results). The effect of these 2 would be a potentially quite large performance hitch as a result of doing this.

None of these are insurmountable problems that mean this approach is doomed to failure, but just things to bear in mind.

If you choose to use immediate mode, you don't strictly need to duplicate everything in immediate mode.

You can find the set of interesting shapes by performing an overlap query against the scene with the bounds of your vehicle.

You can then perform contact generation between the geometries from those shapes and the vehicle's geometries, producing a set of contacts to solve. You won't need to create new shapes/geometries - you can just extract them from the actors and use them directly. The contact gen only requires a pair of geometries and a pair of transforms - there is no link between bodies and geometries in immediate mode (you handle that however works best from the data you have available).

To solve the contacts, you should only need to create 2 bodies - the vehicle's body and a static "dummy" body(with identity pose) to represent all the immoveable objects in your world. You then create the constraints from the contacts between these 2 bodies. As long as the other bodies are to be treated like infinite mass bodies with zero velocity, you should not need to replicate a body for them.

If you do need to factor in their velocities or simulate a 2-way collision response between the bodies, you would need to create a representation for each of those bodies in immediate mode.

Hope this helps

Kier

StormtideGames, 2019-03-05:
Thanks Kier, that's actually really helpful!

I had not considered the cost of reinsertion and loss of cached information. I suppose my other potential performance drain would be the constant locking of the scene for reads/writes too. I'll take the path of immediatte mode for now, and see if that's more scalable.

Kind Regards!

closed 2019-03-05 by StormtideGames