NVIDIAGameWorks/PhysX #80: Checked configuration CRT linkage problem with vcpkg

RT222, 2019-03-10:
Hello,

I'm working on a [Physx port to vcpkg](https://github.com/Microsoft/vcpkg/pull/5618). Vcpkg only support two different configuration modes : Release and Debug. For the debug configuration, I'd like to use the Physx checked configuration mode, because it's what the documentation seems to recommend. But libraries build in debug configuration in vcpkg must use /MDd, and the Physx checked configuration is using /MD.

I know something similar has already been asked [here](https://github.com/NVIDIAGameWorks/PhysX/issues/29), but I wanted to extend the question : can't the checked configuration simply be built with /MDd? I tried it manually on the solution generated by cmake, and it seemed to work fine.

As I'm not an expert on the linking topic, and given that Physx is a pretty complex beast, I'm probably missing something. but I wanted to be sure. But if it's possible without any drawback, it would be awesome if you could add a new flag in the preset files to allow the building of the checked configuration with /MDd.

AlesBorovicka, 2019-03-11:
Hi,
 yes, it is recommended to use checked configuration, but having said that using checked configuration in debug configurations with debug runtime and DEBUG define will be essentially the same as using PhysX debug with enabled optimizations. 
Checked configuration was meant to be as fast as possible, but still providing enough checking to inform users that something is wrong. It is true that mostly this configuration would be used/recommended for development, where in large projects typically additional configuration (like Develop) exists. 
So for DEBUG I would probably recommend to use debug configuration of PhysX.
Ales

RT222, 2019-03-11:
I thought there would be an important performance gap between the checked configuration with /MDd and the debug configuration. But if it's more or less the same, using the later for the vcpkg debug configuration is effectively the best option. Thanks for the fast reply.

closed 2019-03-11 by RT222

AlesBorovicka, 2019-03-11:
There is a gap between those two, note that I wrote that it would match debug config with optimizations enabled. There will be a large performance difference.

RT222, 2019-03-11:
Ah you're right, I missed that part. Thanks for the clarification.