NVIDIAGameWorks/PhysX #89: Batched queries

leijiwei1984, 2019-03-26:
Hi
There is a sentence in the document for Physx 3.x.
"The batched query feature has been deprecated in PhysX version 3.4."
Does it works in physx 4.x?
If I want to use the similar function, what should I do?
Thank you!

AlesBorovicka, 2019-03-26:
Hi,
 Yes, batched queries were marked as deprecated, since the original purpose was to run the queries efficiently on ps3 SPUs. The current implementation does not provide any performance benefits and we would like to replace them with something better. Having said that, we are still considering options here, so batched queries will most likely still be present for some time in current state.
Regards,
Ales

leijiwei1984, 2019-03-26:
Hi,
Thanks.
I inputed physx::PxQueryCache when called PxScene::raycast method.In source codes.
I'm sure the data[actor shape faceindex] of the cache is valid.But i did't find the faceindex used.

leijiwei1984, 2019-03-26:
In my physics scene, there was only one trianglemesh.
First time, I got the hit information when called PxScene::raycast. Inner the SDK's code,it called raycast_trianglemesh.
Second time,I called PxScene::raycast and inputed a valid PxQueryCache. Inner the SDK's code, it called raycast_trianglemesh again. It seemed that the faceindex was not used.
The version of the SDK is Physx3.3.3.

AlesBorovicka, 2019-03-26:
I dont know the history here, but the API doc says:
PxU32			faceIndex;		//!< Triangle index to test first - NOT CURRENTLY SUPPORTED
So it is not supported atm

closed 2021-04-09 by PierreTerdiman