NVIDIAGameWorks/PhysX #92: PhysX 4.0 link dependecies (Linux)

trajcenikolovnick, 2019-04-01:
Hi,

I have built and ran successfully the SDK. I am trying now to integrate into our simulation software. The SDK static libs are linked ok as well as cuda, still lots of messages, most of them are pointing to the: physx::Cm::, physx::profile:: etc.

Where can I get the full list of all the dependencies to link it properly?

Thank you

AlesBorovicka, 2019-04-01:
Hi,
 this is the linking order that should work (it has been updated like this in 4.1 snippets cmake files):
PhysX PhysXPvdSDK PhysXVehicle PhysXCharacterKinematic PhysXExtensions PhysXCooking PhysXCommon PhysXFoundation

You might not need the vehicle/cct libraries. 

Regards,
Ales

trajcenikolovnick, 2019-04-01:
yes. it works .. thanks a bunch!

trajcenikolovnick, 2019-04-02:
Hi again,

it was working this way until I added PxInitExtensions(*gPhysics, gPvd); in the code. Now I am facing lots of undefines from the extension library... Any clue, hint?

Thank you

AlesBorovicka, 2019-04-02:
The PhysX extensions code is in this library libPhysXExtensions_static_64.a make sure you link it in the correct order.

You can also check https://github.com/NVIDIAGameWorks/PhysX/issues/53

trajcenikolovnick, 2019-04-02:
yes, I figured that out ... This is the order and the libs I am linking against. Still lots of undefines, all from SnRepXCoreSerializer.cpp

-lPhysX_static_64 -lPhysXExtensions_static_64 -lPhysXPvdSDK_static_64 -lPhysXVehicle_static_64 -lPhysXCharacterKinematic_static_64  -lPhysXCooking_static_64 -lPhysXCommon_static_64 -lPhysXFoundation_static_64

AlesBorovicka, 2019-04-02:
So on linux:
Linker works from left to right, so put all your dependent libraries on the left, and the ones that satisfy those dependencies on the right of the link line

Could you try to actually put PhysXExtensions before PhysX?

trajcenikolovnick, 2019-04-02:
yeah .. that solved it :) .. Thanks again

closed 2019-04-02 by trajcenikolovnick