NVIDIAGameWorks/PhysX #94: vehicle dynamics and friction pairs (probably low friction skidding)

trajcenikolovnick, 2019-04-03:
Hi,

I have started with mimic a snippet code into our target simulation software - early stage but looks promising.  All works well with the snippet and my mods there. However, in our software when I connect physX with our dynamics engine, I am experiencing some sort of 'skidding' movement of the vehicle (6 wheeled truck). In the vehicle snippets I am seeing tire and surface types defined but not used (simple grep). Any hints of how to drive with normal tires on tarmac surface - or anything else that will allow me to drive the vehicle normally without skidding like on ice - any hints, pointers, links are welcome

Thanks a bunch as always!

Cheers,
Nick

gyeomannvidia, 2019-04-04:
The tire and surface types are used in the vehicle snippets.  They are combined in the function createFrictionPairs and then passed to PxVehicleUpdates as a function parameter.  The system works by associating PxMaterial pointers with surface types and then using a 2d lookup table to compute a friction value for a combined tyre and surface type.  

Hope this helps,

Gordon

trajcenikolovnick, 2019-04-08:
Hi Gordon,

thanks for the explanation. Reading the code I can see now how the material is used along with the 2d lookup table.

I mimic the code from the snippets to get SLICKS tire types on TARMAC surface and the attached video is the result. Bellow is the snippet. What are the settings to make this work better, without skidding?

`#if 0
static PxF32 g_tireFrictionMultipliers[MAX_NUM_SURFACE_TYPES][MAX_NUM_TIRE_TYPES] =
{
    //WETS    SLICKS    ICE        MUD
    {1.70f,    1.85f,    1.70f,    1.70f},       //MUD
    {2.00f,    200.5f,    1.90f,    1.90f},       //TARMAC
    {0.70f,    0.70f,    1.20f,    0.90f},       //ICE
    {1.20f,    0.90f,    1.30f,    1.40f}        //GRASS
};
#else
static PxF32 g_tireFrictionMultipliers[MAX_NUM_SURFACE_TYPES][MAX_NUM_TIRE_TYPES] =
{
    //WETS    SLICKS    ICE        MUD
    {200.5f,    200.5f,    200.5f,    200.5f},       //MUD
    {200.5f,    200.5f,    200.5f,    200.5f},       //TARMAC
    {200.5f,    200.5f,    200.5f,    200.5f},       //ICE
    {200.5f,    200.5f,    200.5f,    200.5f}        //GRASS
};
#endif

PxVehicleDrivableSurfaceToTireFrictionPairs* createFrictionPairs(PxMaterial** defaultMaterial)
{        
    for ( int i=0; i<MAX_NUM_SURFACE_TYPES; ++i )
    {
        _surfaceMaterials[i] = gPhysics->createMaterial( 200.5f, 200.5f, 0.1f );
        _surfaceTypes[i].mType = (PxU32)SURFACE_MUD + i;
    }

    _surfaceTirePairs = PxVehicleDrivableSurfaceToTireFrictionPairs::allocate( MAX_NUM_TIRE_TYPES, MAX_NUM_SURFACE_TYPES );
    _surfaceTirePairs->setup( MAX_NUM_TIRE_TYPES, MAX_NUM_SURFACE_TYPES,
                              (const PxMaterial**)_surfaceMaterials, _surfaceTypes );

    for ( int i=0; i<MAX_NUM_SURFACE_TYPES; ++i )
    {
        for ( int j=0; j<MAX_NUM_TIRE_TYPES; ++j )
            _surfaceTirePairs->setTypePairFriction( i, j, g_tireFrictionMultipliers[i][j] );
    }

    *defaultMaterial = _surfaceMaterials[SURFACE_TARMAC];

    return _surfaceTirePairs;
}`

[physX.mp4.zip](https://github.com/NVIDIAGameWorks/PhysX/files/3052784/physX.mp4.zip)

gyeomannvidia, 2019-04-15:
It looks like the wheels are spinning around way too fast. 

It's very hard to debug without access to code but you want to either increase the moment of inertia of the wheels or decrease the engine maximum torque or filter the throttle control values.

trajcenikolovnick, 2019-04-16:
Hi,

that helped a lot. Thanks for the hints

closed 2019-04-16 by trajcenikolovnick

rockyyliang, 2021-02-07:
@trajcenikolovnick any notes on how you solved this? I flipped the coordinate system by setting new basis vectors and rewriting snippet functions for Z up and now it's like cars are on ice