NVIDIAGameWorks/PhysX #95: PVD crashes on hello world snippet

jbeard4, 2019-04-04:
Physx 4.1, PVD 3.2018.04.23896843, on Windows Server 2019

Steps to reproduce:
1. Start PVD
2. Run Hello world snippet in visual studio

What should happen:
Hello World should attach to the PVD example, as described in the [documentation](http://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/HelloWorld.html#using-physx-visual-debugger-with-snippethelloworld).

What actually happens:
Hello world runs successfully, and renders in glut, but PVD crashes.

Stack trace:

```
==============================================================================
   A Program Fault occurred:
   Error code 0xC0000005: EXCEPTION_ACCESS_VIOLATION
   Address: 0x00007FFAACB05BAD
   Flags: 0x00000000
==============================================================================
==                             STACK TRACE                                  ==
==============================================================================
function: QOpenGLFunctions_4_1_Core::glGenVertexArrays at 0x7FFAACB05BAD
		***  0 called from 0x00007FFAACB05BAD  STACK 0x0000000E5DCFAC78   AddrReturn 0x00007FF61BABE758
function: OpenGLWidget::addArrowGeometry at 0x7FF61BABE758
		***  1 called from 0x00007FF61BABE758  STACK 0x0000000E5DCFAC80   AddrReturn 0x00007FF61BAFB563
function: PVDRenderer::initialize at 0x7FF61BAFB563
		***  2 called from 0x00007FF61BAFB563  STACK 0x0000000E5DCFAD50   AddrReturn 0x00007FF61BAA7ABC
function: PVDMainWindow::initClipFromNet at 0x7FF61BAA7ABC
		***  3 called from 0x00007FF61BAA7ABC  STACK 0x0000000E5DCFAD80   AddrReturn 0x00007FF61BB05992
function: PVDMainWindow::qt_static_metacall at 0x7FF61BB05992
		***  4 called from 0x00007FF61BB05992  STACK 0x0000000E5DCFAE30   AddrReturn 0x0000000054D450F8
function: QObject::event at 0x54D450F8
		***  5 called from 0x0000000054D450F8  STACK 0x0000000E5DCFAF30   AddrReturn 0x000000005463C7FF
function: QWidget::event at 0x5463C7FF
		***  6 called from 0x000000005463C7FF  STACK 0x0000000E5DCFB0D0   AddrReturn 0x00000000546185A0
function: QApplicationPrivate::notify_helper at 0x546185A0
		***  7 called from 0x00000000546185A0  STACK 0x0000000E5DCFB2A0   AddrReturn 0x0000000054617557
function: QApplication::notify at 0x54617557
		***  8 called from 0x0000000054617557  STACK 0x0000000E5DCFB2D0   AddrReturn 0x0000000054D1FB89
function: QCoreApplication::notifyInternal2 at 0x54D1FB89
		***  9 called from 0x0000000054D1FB89  STACK 0x0000000E5DCFB9C0   AddrReturn 0x0000000054D21A71
function: QCoreApplicationPrivate::sendPostedEvents at 0x54D21A71
		*** 10 called from 0x0000000054D21A71  STACK 0x0000000E5DCFBA40   AddrReturn 0x00007FFAAEA3D9EF
function: qt_plugin_query_metadata at 0x7FFAAEA3D9EF
		*** 11 called from 0x00007FFAAEA3D9EF  STACK 0x0000000E5DCFBB10   AddrReturn 0x0000000054D68336
function: QEventDispatcherWin32::processEvents at 0x54D68336
		*** 12 called from 0x0000000054D68336  STACK 0x0000000E5DCFBB40   AddrReturn 0x00007FFAC1BFCA66
function: DispatchMessageW at 0x7FFAC1BFCA66
		*** 13 called from 0x00007FFAC1BFCA66  STACK 0x0000000E5DCFBC90   AddrReturn 0x00007FFAC1BFC582
function: DispatchMessageW at 0x7FFAC1BFC582
		*** 14 called from 0x00007FFAC1BFC582  STACK 0x0000000E5DCFBE10   AddrReturn 0x0000000054D67B00
function: QEventDispatcherWin32::processEvents at 0x54D67B00
		*** 15 called from 0x0000000054D67B00  STACK 0x0000000E5DCFBE90   AddrReturn 0x00007FFAAEA3D9C9
function: qt_plugin_query_metadata at 0x7FFAAEA3D9C9
		*** 16 called from 0x00007FFAAEA3D9C9  STACK 0x0000000E5DCFF000   AddrReturn 0x0000000054D1B99B
function: QEventLoop::exec at 0x54D1B99B
		*** 17 called from 0x0000000054D1B99B  STACK 0x0000000E5DCFF030   AddrReturn 0x0000000054D1EA21
function: QCoreApplication::exec at 0x54D1EA21
		*** 18 called from 0x0000000054D1EA21  STACK 0x0000000E5DCFF0B0   AddrReturn 0x00007FF61BA9020D
function: main at 0x7FF61BA9020D
		*** 19 called from 0x00007FF61BA9020D  STACK 0x0000000E5DCFF120   AddrReturn 0x00007FF61BB78A04
function: WinMain at 0x7FF61BB78A04
	filename: c:\users\qt\work\qt\qtbase\src\winmain\qtmain_win.cpp, line 104
		*** 20 called from 0x00007FF61BB78A04  STACK 0x0000000E5DCFF720   AddrReturn 0x00007FF61BB77C0E
function: __scrt_common_main_seh at 0x7FF61BB77C0E
	filename: f:\dd\vctools\crt\vcstartup\src\startup\exe_common.inl, line 283
		*** 21 called from 0x00007FF61BB77C0E  STACK 0x0000000E5DCFF7B0   AddrReturn 0x00007FFAC23081F4
function: BaseThreadInitThunk at 0x7FFAC23081F4
		*** 22 called from 0x00007FFAC23081F4  STACK 0x0000000E5DCFF7F0   AddrReturn 0x00007FFAC294A251
function: RtlUserThreadStart at 0x7FFAC294A251
		*** 23 called from 0x00007FFAC294A251  STACK 0x0000000E5DCFF820   AddrReturn 0x0000000000000000
function: RtlUserThreadStart at 0x7FFAC294A251
		*** 24 called from 0x00007FFAC294A251  STACK 0x0000000E5DCFF820   AddrReturn 0x0000000000000000

```

jbeard4, 2019-04-04:
I can provide access to an AMI if that would be helpful to reproduce the issue.

jbeard4, 2019-04-04:
I also tried the PVD file approach (as opposed to the network approach), and when I opened the file in the PVD debugger, it returned the error box which said "PVD file is invalid"

I installed the PVD version from [here](https://developer.nvidia.com/physx-visual-debugger), according to the PhysX 4.1 documentation [here](http://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/VisualDebugger.html#physxvisualdebugger). I have a suspicion that this version might be out of date. Do I need to build an up-to-date version of PVD from source?

I would appreciate any insight you can offer. Thank you.

PierreTerdiman, 2019-04-04:
Do you know which OpenGL version is supported on your device? That callstack ends in glGenVertexArrays, which is an OpenGL 3.0 function (https://www.khronos.org/registry/OpenGL-Refpages/gl4/html/glGenVertexArrays.xhtml)

PierreTerdiman, 2019-04-04:
About that "PVD file is invalid" message: it might be because it has an incorrect / unexpected extension. Could you try to rename it to a ".pxd2" extension if it doesn't have it already?

jbeard4, 2019-04-06:
Hi, thank you for your insight on this issue.

The issue I was encountering was described [here](https://www.reddit.com/r/cloudygamer/comments/7twmiq/problem_with_azure_nv6_and_nvidia_tesla_m60/) and  [here](https://forums.aws.amazon.com/thread.jspa?messageID=500885). Basically, it was using the generic Microsoft Display adapter, which supports OpenGL 1.1 only.

The resolution was described [here](https://www.reddit.com/r/cloudygamer/comments/7twmiq/problem_with_azure_nv6_and_nvidia_tesla_m60/dtjpld2?utm_source=share&utm_medium=web2x) and [here](https://docs.nvidia.com/gameworks/content/developertools/desktop/nsight/tesla_compute_cluster.htm), which describes how to enable WDDM mode on the nVidia Tesla. The nVidia driver now supports OpenGL 4.4 (as reported by the OpenGL Extension Viewer 5.2). 

PVD is now working great. Thank you again for your help with this.

closed 2019-04-06 by jbeard4

DrCorvax, 2019-08-21:
I've got similar problem - PVD crashes with Hello Word snippet, but error is different.
OS is Windows 10 ver. 18632.295.
PhysX is built with Visual Studio 2019 ver. 16.2.3.
PVD is ver. 3.2019.04.26214843.
[9089325c-4065-4a41-aff4-766ff84f4664_stack.txt](https://github.com/NVIDIAGameWorks/PhysX/files/3525302/9089325c-4065-4a41-aff4-766ff84f4664_stack.txt)

cm56uw, 2023-07-19:
> I've got similar problem - PVD crashes with Hello Word snippet, but error is different. OS is Windows 10 ver. 18632.295. PhysX is built with Visual Studio 2019 ver. 16.2.3. PVD is ver. [...]

Hello, I had the same problem with the same stack log, do you find any solution? Thank you.

preist-nvidia, 2023-07-19:
Hi @cm56uw - PVD is deprecated, but we are actively developing its replacement OmniPVD: https://docs.omniverse.nvidia.com/extensions/latest/ext_omnipvd.html