NVIDIAGameWorks/PhysX #100: setKinematicSurfaceVelocity missing

alektron, 2019-04-09:
In the release notes for 3.4.0.21821222  the addition of the rigid body function/feature "setKinematicSurfaceVelocity" is stated.

It is also mentioned in the User Guide for 4.0 and 4.1 (not in 3.4 though).
This functions is nowhere to be found. As it seems not in 3.4 branch as well as not in 4.0 or 4.1

alektron, 2019-04-23:
I now implemented the needed functionality myself by using contact modification and target velocity.

Maybe the paragraph in the User Guide should be removed. Or the function should be added again. Either way, the way it is now might be confusing to others as well

wanogr, 2020-02-03:
I also need this function. maybe it will appear in version 5?