NVIDIAGameWorks/PhysX #101: How to get all the hits from capsule/triangleMesh (concave) collision

chico19, 2019-04-09:
Hi,

I'm trying to get all the collision points from the interaction of a dynamic capsule against a triangle mesh (concave shape). The dynamic capsule is controlled by a kinematic capsule through a D6Joint.  The transform of the kinematic capsule is set in real time by a haptic device. 

I tried to use scene query sweep to get the collision information. At each time step, I set the dynamic capsule as the sweep object, the direction to the kinematic capsule as the sweep direction. As far as I know, the sweep can only return at most 1 hit and eMESH_MULTIPLE hit flag is not supported for sweep. So I cannot get all the hits when there are more than 1 hit between the capsule and the concave mesh. Even though it says the scene query sweep can return multiple hits if you set your own PxSweepBuffer, I guess it means multiple hits for different objects, but still one hit for each object?

Also, when the capsule is consistently touching the mesh, the returned position and distance can be undefined if  PxHitFlag::eMTD is not set, which I think is because the capsule and the mesh can be initially overlapped during consistent touch and if eMTD is not set, position and distance will be invalid as stated in the documentation.

So, my questions are:
1. Is there a way to get all the collision points?
2. How to calculate the overall force the capsule applied to the mesh so that I can pass to the haptic device to simulate the force feedback?

Thanks in advance!

AlesBorovicka, 2019-04-11:
Hi,
 no we dont support sweep mesh multiple, you can get only the closest face with sweeps against triangle mesh. I guess you could get the contact points through a simulation contact stream? Would that not work for you?

Ales

chico19, 2019-04-24:
> Hi,
> no we dont support sweep mesh multiple, you can get only the closest face with sweeps against triangle mesh. I guess you could get the contact points through a simulation contact stream? Would that not work for you?
> 
> Ales

Hi Ales,
Thanks for your reply.  Do you mean onContact in simulationEventCallback? I have tried that, it can return the contact information between the mesh and the dynamic capsule. But I have some questions:

1. The onContact callback returns contact positions even before the capsule is actually touching the mesh, when the capsule is very close to the mesh but still not touching, the onContact callback already returns some contact points. Is there a collision margin or something?

2. The contact separation can be positive sometimes, but in the doc, it says, the separation denotes a penetration, they are all negative, no positive.

3. At one contact location, there are usually many contact points, but I only need one contact point, should I pick the one with largest penetration? 

4. Since the dynamic capsule is controlled by the kinematic capsule through a Dx6joint, the contact information returned can be unstable when the separation of the two capsules becomes large. I can see some jitter and instability. Is there a way to avoid that?

Thanks again!

pkxpp, 2023-04-03:
*PxMeshQuery::findOverlapTriangleMesh* can be used

```
PxU32 triangleIndexBuffer[bufferSize];
PxU32 startIndex = 0;
bool bufferOverflowOccured = false;
PxU32 nbTriangles = PxMeshQuery::findOverlapTriangleMesh(sphereGeom, spherePose,
                                                         meshGeom, meshPose,
                                                         triangleIndexBuffer, bufferSize,
                                                         startIndex, bufferOverflowOccured);

for(PxU32 i=0; i < nbTriangles; i++)
{
    PxTriangle tri;
    PxU32 vertexIndices[3];
    PxMeshQuery::getTriangle(meshGeom, meshPose, triangleIndexBuffer[i], tri, vertexIndices);

     ...  // process triangle info
}
```

ref: [findOverlapTriangleMesh](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/GeometryQueries.html)