NVIDIAGameWorks/PhysX #99: Can Blast and NvCloth Use when Migrating From PhysX SDK 3.4 to 4.0

Vovery, 2019-04-09:
We use the Blast and NvCloth depend the Phsyx SDK 3.4 
If we update the physx SDK to 4.1, dose the balst nvcloth can be work as well as before？

nvidia-bgaldrikian, 2019-04-09:
Hi, I can't speak for NvCloth, but since Blast is independent of physics engine, it only relies on some high-level API features which should be quite well behaved in almost any version of PhysX.  PhysX has been improved with each version, becoming faster and more stable (including the GPU solver).  So it should only show improvement, if any change at all, with 4.1.  Of course, there's always a chance that some subtle change will lead to some strange behavior.  But I'm quite sure that could be corrected easily if that did happen to be the case.

Bryan Galdrikian

nvidia-bgaldrikian, 2019-04-09:
There's a decent chance, however, that some API change will cause your physx/Blast integration to not compile.  The code changes required to fix it should be straightforward.

Vovery, 2019-04-18:
> There's a decent chance, however, that some API change will cause your physx/Blast integration to not compile. The code changes required to fix it should be straightforward.

Thanks . I had tried building blast nvcloth dependent the physx 4.1 suscessfully!

closed 2019-04-18 by Vovery