NVIDIAGameWorks/PhysX #102: LinearVelocity is zero of PxArticulationLink

xgerrmann, 2019-04-10:
Hello, 

I am encountering some unexpected behaviour regarding the linear velocity of a PxArticulationLink.

Currently I am reeling in a cable which can be seen in the \*.gif below. I have logged the final position and velocity of the last element. The position seems to change as expected (plot 1). I compared the velocity that I derived from the position (plot 2) and the velocity that I obtained through: PxArticulationLink*->getLinearVelocity() (plot 3).

The reeling is performed by changing the distance of a spherical joint between the last (topmost) articulation link and an attachment.

Why is the linearvelocity (plot 3) of the final element so different from what I expect (plot 2)?

p.s. I do not make use of drives, since these will result in large joint errors due to large mass ratios in the final application. The prescribed reeling velocity must be followed quite precisely.

[image]
[image]
[image]
[image]

kstorey-nvidia, 2019-04-12:
Which solver are you using?

There are a couple of potential reasons this might be happening. 

One might be because you are adjusting the distance on a distance joint. This has the effect of introducing error into the system, which has to be corrected by the solver. A bias is added to the constraints in the positional iterations to correct this error. The bias is zeroed after positional iterations when the velocity iterations are run..

Positional iterations provide the change in poses that are used to integrate the poses of the rigid bodies, whereas velocity iterations produce the velocities that are reported by the rigid bodies and carried over next frame.

The intention of this kind of split impulse approach is to correct error while minimising the energy that is introduced into the simulation to correct the error. In this particular case of reeling in your line, you actually want this velocity to be carried over but, because it is being driven by error, it may/may not leak into the final velocities (depending on how many velocity iterations you use, some of the velocity added to correct error may still be left in the velocity that is reported at the end of the solver.

Retaining/zeroing the bias is controlled by a flag that is raised on the constraint in the joint shader. The PxDistanceJoint shader raises this flag (so bias will not be zeroed) but the D6Joint emulating a distance joint does not raise this flag unless the joint starts off in a non-error state. You could work around this with the D6 joint (if it is the cause) by using a soft spring limit for the distance joint - if you do this, then the bias term is maintained.

The other (quite likely now that I think about it) possibility is that the legacy articulations perform a re-projection step between position and velocity iterations. This projection step corrects internal joint error that has been introduced by the solver but it also tends to remove quite a bit of velocity from the system, This does not happen with the reduced coordinate articulations so this may be the cause of the issue you are seeing. If that doesn't

xgerrmann, 2019-04-16:
Thank you for your elaborate answer,

I have tried both solvers and both give similar results, so a mismatch between the derived velocity and the queried velocity. Furthermore, adjusting the velocity and position solver iterations has no effect whatsoever.

I have not tried the D6 joint approach since I will now continue with the reduced coordinate articulation.

Since the behavior differs between the reduced coordinate articulation and the regular articulation, it seems your final suggestion might explain what is happening.

Currently, I am trying to perform the reeling with a spherical (driven) joint and adding/removing elements to/of the reduced coordinate articulation.

Thanks again for your insights.