NVIDIAGameWorks/PhysX #106: collision of vehicle against vertical terrain surfaces results with vehicle spinning and flying in the air

trajcenikolovnick, 2019-04-16:
Hi,

The code is based on the SnippetVehicleContactMod and was discussed in issue #83

trajcenikolovnick, 2019-04-16:
Here is what we are getting on collision. Is this the same issue or it is something else? Any hints where to look at for resolving it?

[https://www.dropbox.com/s/ifc9uxrmoiakp51/phys4-collision-1-2019-04-16_12.42.53.mp4?dl=0](url)

kstorey-nvidia, 2019-04-16:
A guess - I'd say that the collision is not generating contacts between the vehicle and the side wall when you drive into it because the wall surface is pointing in the wrong direction (if its a mesh, your winding order is incorrect; if it's a terrain, the rotation you used is incorrect). When the vehicle gets sufficiently far through the surface that the COM of its shape is not back-face culled during contact gen, you get a large impulse from the solver to try and force the vehicle through the wall and then the simulation becomes unstable and launches the vehicle into the air.

trajcenikolovnick, 2019-04-16:
Thanks a bunch, this was fast! Your guess is probably right. I did a fast test with backface culling rendering of the terrain and indeed I am seeing non-consistency  of the winding order for faces that are part of the walls. I am closing this issue and will give the terrain back to the artists

Thank you again

closed 2019-04-16 by trajcenikolovnick