NVIDIAGameWorks/PhysX #107: No CMAKE_CXX_COMPILER could be found

padalous, 2019-04-17:
I got the following error when running `./generate_projects.sh`:
`Preset parameter required, available presets:
(0) ios64 <--- iOS Xcode PhysX general settings
(1) mac64 <--- macOS Xcode PhysX general settings
(2) linux <--- Linux clang PhysX SDK general settings
(3) linux-aarch64 <--- Linux-aarch64 gcc PhysX SDK general settings
Enter preset number: 1
Running generate_projects.bat mac64
Using preset xml: buildtools/presets/public/mac64.xml
Target platform: mac64 using compiler: xcode
PM_CMakeModules_PATH: /Users/paul.dalous/PhysX/physx/../externals/cmakemodules
PM_PATHS: /Users/paul.dalous/PhysX/physx/../externals/opengl-linux
Cmake: cmake
Not searching for unused variables given on the command line.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:34 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:34 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "/Users/paul.dalous/PhysX/physx/compiler/mac64/CMakeFiles/CMakeOutput.log".
See also "/Users/paul.dalous/PhysX/physx/compiler/mac64/CMakeFiles/CMakeError.log".`

AlesBorovicka, 2019-04-17:
Looking at the log, cmake tries to create xcode projects and the compiler check fails.

Make sure you have xcode installed.

padalous, 2019-04-17:
Yes, Xcode 10.2 is installed

padalous, 2019-04-17:
Solved by following : https://stackoverflow.com/questions/41380900/cmake-error-no-cmake-c-compiler-could-be-found-using-xcode-and-glfw

It was indeed, an Xcode issue, I had to do `sudo xcode-select --reset`

Thank you for your help @AlesBorovicka

PcChip, 2024-04-28:
Hello,

I'm getting a simliar error on Kubuntu 23.10:  
  
````
mike@ALDERLAKE:~/Documents/PhysX/physx$ ./generate_projects.sh 
Preset parameter required, available presets:
(0) linux-aarch64 <--- Linux-aarch64 gcc PhysX SDK general settings
(1) linux <--- Linux clang PhysX SDK general settings
Enter preset number: 1
Running generate_projects.sh linux
Running packman in preparation for cmake ...
Using preset xml: buildtools/presets/public/linux.xml
Target platform: linux using compiler: clang
PM_PATHS: /home/mike/.cache/packman/chk/clang-physxmetadata/4.0.0.32489833_1;/home/mike/.cache/packman/chk/PhysXGpu/105.1-5.3.1.6193-f2687f97-linux-x86_64-public;/home/mike/.cache/packman/chk/opengl-linux/2017.5.19.1;/home/mike/.cache/packman/chk/rapidjson/1.1.0-67fac85-073453e1
Cmake: cmake
Not searching for unused variables given on the command line.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:38 (project):
  The CMAKE_C_COMPILER:

    clang

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

````

ayoub-belarbi, 2024-04-29:
Hello @PcChip,
Do you have clang/clang++ installed?
Thanks

PcChip, 2024-04-29:
> Hello @PcChip, Do you have clang/clang++ installed? Thanks

hello, that was the issue, thank you for your reply!

closed 2024-04-29 by preist-nvidia