NVIDIAGameWorks/PhysX #108: Empty (zero) cache after adding a new link

xgerrmann, 2019-04-19:
I am comparing the positions, velocities, and accelerations of all joints of a reduced coordinate articulation of an articulation before and after removing and adding a link.

A problem that I am running into is that the cache after **removing one link and adding two links** is completely different from the one before it.

**###############**
**Cache Before**
Link: 00, #DOF: 0
Link: 01, #DOF: 3
DOF#: 00, pos: -0.147797, 	 vel: 1.586147, 	 acc: 0.763219
DOF#: 01, pos: 0.010139, 	 vel: 0.002495, 	 acc: -0.001043
DOF#: 02, pos: 0.146445, 	 vel: -1.607627, 	 acc: -0.752921
Link: 02, #DOF: 3
DOF#: 00, pos: -0.006484, 	 vel: -3.409779, 	 acc: -0.142321
DOF#: 01, pos: -0.000834, 	 vel: -0.000123, 	 acc: -0.004698
DOF#: 02, pos: 0.007750, 	 vel: 3.457763, 	 acc: 0.122109
Link: 03, #DOF: 1
DOF#: 00, pos: 0.001277, 	 vel: 0.048320, 	 acc: 10.055427

**Cache After**
Link: 00, #DOF: 0
Link: 01, #DOF: 3
DOF#: 00, pos: 0.000000, 	 vel: 0.000000, 	 acc: 0.000000
DOF#: 01, pos: 0.000000, 	 vel: 0.000000, 	 acc: 0.000000
DOF#: 02, pos: 0.000000, 	 vel: 0.000000, 	 acc: 0.000000
Link: 02, #DOF: 3
DOF#: 00, pos: 0.000000, 	 vel: 0.000000, 	 acc: 0.000000
DOF#: 01, pos: 0.000000, 	 vel: 0.000000, 	 acc: 0.000000
DOF#: 02, pos: 0.000000, 	 vel: 0.000000, 	 acc: 0.000000
Link: 03, #DOF: -1
Error in DOF.

Link: 04, #DOF: -1
Error in DOF.
**###############**

I made it print an error ('Error in DOF') when it thinks there are more than 3 DOF.
Here is the [github Gist](https://gist.github.com/xgerrmann/e02fd0d0f8938912a1db94ec3f61271c) that makes it print the articulation.

In the [documentation](http://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Articulations.html?highlight=pxarticulationcache#pxarticulationcache) it says:
> if the properties of an articulation change (a link is added/removed, degrees of freedom are changed etc.), it is necessary to destroy the cache and recreate it.

Therefore I have tried 2 options:
* Releasing the Cache and creating a new cache after the link removal and addition before printing :
`gArticulation->releaseCache(*articulationCache);`
`articulationCache = gArticulation->createCache();`
This leads to the above two caches.
* NOT releasing the Cache and NOT creating a new cache after the link removal and addition. This leads to thw two caches below

The caches are printed directly before and after the change in the articulation, no simulation step is performed.

**###############**
**Cache Before**
Link: 00, #DOF: 0
Link: 01, #DOF: 3
DOF#: 00, pos: -0.181617, 	 vel: 2.423691, 	 acc: 1.930781
DOF#: 01, pos: 0.015657, 	 vel: 0.006166, 	 acc: 0.000054
DOF#: 02, pos: 0.179895, 	 vel: -2.449120, 	 acc: -1.901229
Link: 02, #DOF: 3
DOF#: 00, pos: 0.013653, 	 vel: -4.690832, 	 acc: -1.883253
DOF#: 01, pos: -0.000786, 	 vel: 0.001315, 	 acc: -0.003380
DOF#: 02, pos: -0.013040, 	 vel: 4.747666, 	 acc: 1.794042
Link: 03, #DOF: 1
DOF#: 00, pos: 0.006076, 	 vel: -0.221526, 	 acc: 18.003754

**Cache After**
Link: 00, #DOF: 0
Link: 01, #DOF: 3
DOF#: 00, pos: -0.181617, 	 vel: 2.423691, 	 acc: 1.930781
DOF#: 01, pos: 0.015657, 	 vel: 0.006166, 	 acc: 0.000054
DOF#: 02, pos: 0.179895, 	 vel: -2.449120, 	 acc: -1.901229
Link: 02, #DOF: 3
DOF#: 00, pos: 0.013653, 	 vel: -4.690832, 	 acc: -1.883253
DOF#: 01, pos: -0.000786, 	 vel: 0.001315, 	 acc: -0.003380
DOF#: 02, pos: -0.013040, 	 vel: 4.747666, 	 acc: 1.794042
Link: 03, #DOF: -1
Error in DOF.

Link: 04, #DOF: -1
Error in DOF.
**###############**

In each case I made the following observations / hypotheses:
* In the case where I release the cache and recreate it:
   * The number of DOF of the inbound joints seem to be incorrect
   * All positions, velocities and accelerations are suddenly zero.
* In the case where I to NOT release and NOT recreate the cache:
   * The increase in links is properly detected (3 to 4)
   * The velocities of the previous two links seem to be identical to before removing and adding the links.
   * Also here, the DOF are incorrect. 

Can someone tell me how to properly get the cache data from the articulation after modifying the articulation?

Any help would be very much appreciated.

kstorey-nvidia, 2019-04-29:
Apologies for the late response. I will try to set up a repro this week and investigate

Thanks

Kier

xgerrmann, 2019-04-29:
@kstorey-nvidia Would be great,
looking forward to the results!

xgerrmann, 2019-06-05:
Not relevant any more, closing.

closed 2019-06-05 by xgerrmann